ros2/kilted_tutorial_party #264: Executables in `tlsf_cpp`

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `tlsf_allocator_example`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       ros2 run tlsf_cpp tlsf_allocator_example
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-02:
I have completed the following tests and confirmed everything works as expected.

## Checks

* [x]  **Check tlsf_allocator_example**   

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde