ros2/kilted_tutorial_party #265: Executables in gz vendor packages

tfoote, 2025-05-01:
You will need to install the `desktop-full` or `simulation` variants before running this test.

## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `gz sim -v4 shapes.sdf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       gz sim -v4 shapes.sdf
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

leander-dsouza, 2025-05-03:
Everything works as expected :)

This is the output of the Gazebo window:

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde