ros2/kilted_tutorial_party #267: Executables in gz vendor packages

tfoote, 2025-05-01:
You will need to install the `desktop-full` or `simulation` variants before running this test.

## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Checks
- [ ] **Check `gz sim -v4 shapes.sdf`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       gz sim -v4 shapes.sdf
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-02:
I have completed the following tests and confirmed everything works as expected.

## Checks

* [x]  **Check gz sim -v4 shapes.sdf**   

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde