ros2/kilted_tutorial_party #268: Experimenting with a dummy robot

tfoote, 2025-05-01:
Check the documentation for the 'Experimenting with a dummy robot' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Experimenting with a dummy robot page](https://docs.ros.org/en/kilted/Tutorials/Demos/dummy-robot-demo.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

shardrv, 2025-05-04:
Unable to run rviz2 using the commands in the instructions:

`source ~/ros2_ws/install/setup.bash`
`rviz2`

See the error on the terminal:

```
INFO] [1746324848.771498177] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746324848.771705004] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1746324848.866232094] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/opt/ros/kilted/share/rviz_default_plugins/icons/classes/TF.svg]: Failed to open file
Aborted (core dumped)
```

shardrv, 2025-05-05:
After going through issue https://github.com/ros2/rviz/pull/1455 , upgraded the rviz2 using the command

```
sudo apt-get install ros-kilted-rviz2 
```

Terminal output

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be upgraded:
  ros-kilted-rviz2
1 upgraded, 0 newly installed, 0 to remove and 149 not upgraded.
Need to get 19.8 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://packages.ros.org/ros2-testing/ubuntu noble/main amd64 ros-kilted-rviz2 amd64 15.0.0-1noble.20250502.234300 [19.8 kB]
Fetched 19.8 kB in 0s (50.0 kB/s)           
(Reading database ... 314652 files and directories currently installed.)
Preparing to unpack .../ros-kilted-rviz2_15.0.0-1noble.20250502.234300_amd64.deb ...
Unpacking ros-kilted-rviz2 (15.0.0-1noble.20250502.234300) over (15.0.0-1noble.20250428.215941) ...
Setting up ros-kilted-rviz2 (15.0.0-1noble.20250502.234300) ...

```

Ran the *rviz2* again in a separate terminal (after sourcing the terminal) and see the error again:

```
[INFO] [1746405586.747937448] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746405586.748166603] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1746405586.843710634] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/opt/ros/kilted/share/rviz_default_plugins/icons/classes/TF.svg]: Failed to open file
[1]    15878 IOT instruction (core dumped)  rviz2

```

zamantousif, 2025-05-09:
Hmm, I did not try with `ros-kilted-rviz2` but I went with the docker container approach since it already has the relevant packages installed.

I was able to follow the tutorial and see the result on rviz. :)

[image]

> ## Setup
> 
>     * Rmw: Fastdds
>     * BuildType: Debian
>     * Os: Noble
>     * Chip: Amd64
> 
> > 
> ## Checks
> 
>     * [x]  I was able to follow the documentation.
>     * [x]  The documentation seemed clear to me.
>     * [x]  The documentation didn't have any obvious errors.
> 

I think this issue could be closed as all checks passed on the setup!

ahcorde, 2025-05-09:
Thank you for testing