ros2/kilted_tutorial_party #269: Exporting an URDF File

tfoote, 2025-05-01:
Check the documentation for the 'Exporting an URDF File' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Exporting an URDF File page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/URDF/Exporting-an-URDF-File.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-03:
Interested in !

avanmalleghem, 2025-05-03:
Can be closed once PR merged : https://github.com/ros2/ros2_documentation/pull/5504

closed 2025-05-03 by christophebedard