ros2/kilted_tutorial_party #270: FastDDS Configuration

tfoote, 2025-05-01:
Check the documentation for the 'FastDDS Configuration' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [FastDDS Configuration page](https://docs.ros.org/en/kilted/Tutorials/Advanced/FastDDS-Configuration.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Juliaj, 2025-05-03:
Overall:
- [] I was able to follow the documentation.
- [] The documentation seemed clear to me.
- [] The documentation didn't have any obvious errors.

Potential updates 

➜  ros2_kilted ros2 run sync_async_node_example_cpp SyncAsyncWriter 
[WARN] [1746232594.164250212] [rmw_fastrtps_shared_cpp]: FASTRTPS_DEFAULT_PROFILES_FILE value is used for participant configuration, but it is deprecated and will no longer be supported. Use FASTDDS_DEFAULT_PROFILES_FILE instead.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde