ros2/kilted_tutorial_party #271: Gazebo

tfoote, 2025-05-01:
Check the documentation for the 'Gazebo' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Gazebo page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Simulators/Gazebo/Gazebo.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-02:
## Checks

* [x]  I was able to follow the documentation.
* [x]  The documentation seemed clear to me.
* [x]  The documentation didn't have any obvious errors.

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde