ros2/kilted_tutorial_party #272: Getting Started With Ros2doctor

tfoote, 2025-05-01:
Check the documentation for the 'Getting Started With Ros2doctor' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Getting Started With Ros2doctor page](https://docs.ros.org/en/kilted/Tutorials/Beginner-Client-Libraries/Getting-Started-With-Ros2doctor.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

dheerubhai-101, 2025-05-02:
- [x] I was able to follow the documentation.

- [x] The documentation seemed clear to me.

- [x] The documentation didn't have any obvious errors.

1. Setup check

[image]

2. System check

[image]

3. Full report

```
vscode ➜ /workspaces/kilted_ws $ ros2 doctor --report

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
inet6        : ['::1']
netmask      : 255.0.0.0
device       : lo
flags        : UP,LOOPBACK,RUNNING
mtu          : 65536
inet         : 172.17.0.2
inet4        : ['172.17.0.2']
ether        : d2:6a:36:ec:99:40
netmask      : 255.255.0.0
device       : eth0
flags        : UP,BROADCAST,RUNNING,MULTICAST
mtu          : 1500
broadcast    : 172.17.255.255

   PACKAGE VERSIONS
examples_rclcpp_minimal_client            : latest=0.20.5, local=0.20.5
unique_identifier_msgs                    : latest=2.7.0, local=2.7.0
rmw_zenoh_cpp                             : latest=0.6.0, local=0.6.0
ament_cmake_pytest                        : latest=2.7.3, local=2.7.3
cv_bridge                                 : latest=4.1.0, local=4.1.0
rqt_shell                                 : latest=1.3.1, local=1.3.1
examples_rclcpp_minimal_action_client     : latest=0.20.5, local=0.20.5
rosidl_runtime_c                          : latest=4.9.4, local=4.9.4
topic_monitor                             : latest=0.36.0, local=0.36.0
ament_cmake_cpplint                       : latest=0.19.2, local=0.19.2
rviz_rendering                            : latest=15.0.0, local=15.0.0
rosidl_dynamic_typesupport_fastrtps       : latest=0.4.1, local=0.4.1
sdl2_vendor                               : latest=3.3.0, local=3.3.0
laser_geometry                            : latest=2.10.0, local=2.10.0
urdf                                      : latest=2.12.2, local=2.12.2
tracetools                                : latest=8.6.0, local=8.6.0
ament_cmake_target_dependencies           : latest=2.7.3, local=2.7.3
message_filters                           : latest=7.1.0, local=7.1.0
ros_workspace                             : latest=1.0.3, local=1.0.3
ament_cmake_export_targets                : latest=2.7.3, local=2.7.3
ros_environment                           : latest=4.3.1, local=4.3.1
ament_package                             : latest=0.17.2, local=0.17.2
ament_cmake_export_include_directories    : latest=2.7.3, local=2.7.3
qt_gui_cpp                                : latest=2.9.1, local=2.9.1
demo_nodes_cpp                            : latest=0.36.0, local=0.36.0
uncrustify_vendor                         : latest=3.1.0, local=3.1.0
ros2action                                : latest=0.38.0, local=0.38.0
ros2cli_common_extensions                 : latest=0.4.0, local=0.4.0
rcl_logging_spdlog                        : latest=3.2.2, local=3.2.2
robot_state_publisher                     : latest=3.4.2, local=3.4.2
eigen3_cmake_module                       : latest=0.4.0, local=0.4.0
joy                                       : latest=3.3.0, local=3.3.0
ros2doctor                                : latest=0.38.0, local=0.38.0
[... 290 lines cut ...]
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : /turtle1/cmd_vel
publisher count     : 1
subscriber count    : 1
topic               : /turtle1/color_sensor
publisher count     : 1
subscriber count    : 0
topic               : /turtle1/pose
publisher count     : 1
subscriber count    : 0
topic               : /turtle1/rotate_absolute/_action/feedback
publisher count     : 1
subscriber count    : 1
topic               : /turtle1/rotate_absolute/_action/status
publisher count     : 1
subscriber count    : 1
```

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde