ros2/kilted_tutorial_party #273: Image Tools executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Image Tools](https://github.com/ros2/demos/blob/kilted/image_tools/README.md#run)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-05:
Works for me.

[image]

Log output confirming the middleware

```
[INFO] [1746444091.350892000] [cam2image]: Publishing image #5250
[INFO] [1746444091.396536500] [cam2image]: Publishing image #5251
[INFO] [1746444091.491567600] [cam2image]: Publishing image #5252
[INFO] [1746444091.522475500] [cam2image]: Publishing image #5253
[INFO] [1746444091.523782700] [rclcpp]: signal_handler(signum=2)
[INFO] [1746444091.555054000] [cam2image]: Publishing image #5254
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor --report | grep middleware
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'
middleware name    : rmw_cyclonedds_cpp

(pixi_ros2_rolling) C:\pixi_ws>
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde