ros2/kilted_tutorial_party #274: Image Tools executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Image Tools](https://github.com/ros2/demos/blob/kilted/image_tools/README.md#run)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-04:
Testing this ticket

4-legends, 2025-05-04:
- Setup
```
legends@4legends-ASUS:~$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

legends@4legends-ASUS:~$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted

``` 

[image]

- Logs
```
legends@4legends-ASUS:~$ ros2 run image_tools cam2image
[ WARN:0@1.558] global ./modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[INFO] [1746325462.745658972] [cam2image]: Publishing image #1
[INFO] [1746325462.867861264] [cam2image]: Publishing image #2
[INFO] [1746325462.999953218] [cam2image]: Publishing image #3
[INFO] [1746325463.131951058] [cam2image]: Publishing image #4
[INFO] [1746325463.267925072] [cam2image]: Publishing image #5
[INFO] [1746325463.399897371] [cam2image]: Publishing image #6
[INFO] [1746325463.532057437] [cam2image]: Publishing image #7
[INFO] [1746325463.667926444] [cam2image]: Publishing image #8
[INFO] [1746325463.799956220] [cam2image]: Publishing image #9
[INFO] [1746325463.932057384] [cam2image]: Publishing image #10
[INFO] [1746325464.068045744] [cam2image]: Publishing image #11
[INFO] [1746325464.200046278] [cam2image]: Publishing image #12
[INFO] [1746325464.332006280] [cam2image]: Publishing image #13
c[INFO] [1746325464.467964598] [cam2image]: Publishing image #14
[INFO] [1746325464.600084411] [cam2image]: Publishing image #15
[INFO] [1746325464.731947086] [cam2image]: Publishing image #16
[INFO] [1746325464.863927099] [cam2image]: Publishing image #17
[INFO] [1746325464.999924251] [cam2image]: Publishing image #18
[INFO] [1746325465.131956475] [cam2image]: Publishing image #19
[INFO] [1746325465.263898788] [cam2image]: Publishing image #20
[INFO] [1746325465.399973003] [cam2image]: Publishing image #21
[INFO] [1746325465.532021138] [cam2image]: Publishing image #22
[INFO] [1746325465.664063971] [cam2image]: Publishing image #23
[INFO] [1746325465.799853838] [cam2image]: Publishing image #24
[INFO] [1746325465.931985206] [cam2image]: Publishing image #25
^C[ros2run]: Received signal:  Interrupt
[INFO] [1746325466.063413650] [rclcpp]: signal_handler(signum=2)
[INFO] [1746325466.063907884] [cam2image]: Publishing image #26

``` 

```
legends@4legends-ASUS:~$ ros2 run image_tools showimage
[INFO] [1746325457.967671910] [showimage]: Subscribing to topic 'image'
[INFO] [1746325462.748561066] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325462.971902875] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325463.001102534] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325463.133374149] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325463.269109511] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325463.401175457] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325463.533243877] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325463.669212206] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325463.801115767] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325463.933443847] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325464.069117780] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325464.201349810] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325464.333212464] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325464.469078237] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325464.601215068] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325464.733117886] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325464.864896340] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325465.000929469] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325465.132924907] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325465.264933358] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325465.400784706] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325465.532968191] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325465.665105354] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325465.800759932] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746325465.933012729] [showimage]: Received image #camera_frame
Received image #camera_frame
^C[INFO] [1746325467.239053364] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746325467.239292088] [rclcpp]: signal_handler(signum=2)

```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde