ros2/kilted_tutorial_party #275: Image Tools executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Image Tools](https://github.com/ros2/demos/blob/kilted/image_tools/README.md#run)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

yashphalle, 2025-05-03:
## Executables are working as expected.

I've tested the image_tools executables as described [here ](https://github.com/ros2/demos/blob/kilted/image_tools/README.md#run)  and can confirm they're working correctly in the specified environment (FastDDS/Binary/Windows/AMD64).

Both cam2image and showimage executables function as documented in the README. 
The cam2image successfully publishes camera feeds to the specified topics, and showimage properly displays the images from the subscribed topic.

ahcorde, 2025-05-05:
Do you mind to add a screenshot ?

Aniketkumarroy, 2025-05-13:
Hi I tested this

[image]
### ***My system***

[image]

ahcorde, 2025-05-13:
Thank you for testing

closed 2025-05-13 by ahcorde