ros2/kilted_tutorial_party #276: Image Tools executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Image Tools](https://github.com/ros2/demos/blob/kilted/image_tools/README.md#run)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-03:
Testing this ticket

4-legends, 2025-05-03:
Setup:

```
legends@4legends-ASUS:~/ros2_ws$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted

legends@4legends-ASUS:~/ros2_ws$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_fastrtps_cpp

``` 

Output:

```
legends@4legends-ASUS:~$ ros2 run image_tools cam2image
[ WARN:0@1.016] global ./modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
[INFO] [1746291481.720292324] [cam2image]: Publishing image #1
[INFO] [1746291481.775050247] [cam2image]: Publishing image #2
[INFO] [1746291481.838949767] [cam2image]: Publishing image #3
[INFO] [1746291481.907089686] [cam2image]: Publishing image #4
[INFO] [1746291481.975047599] [cam2image]: Publishing image #5
[INFO] [1746291482.039091955] [cam2image]: Publishing image #6
[INFO] [1746291482.106919494] [cam2image]: Publishing image #7
[INFO] [1746291482.174890282] [cam2image]: Publishing image #8
[INFO] [1746291482.239017162] [cam2image]: Publishing image #9
``` 

```
legends@4legends-ASUS:~$ ros2 run image_tools showimage
[INFO] [1746291655.819162542] [showimage]: Subscribing to topic 'image'
[INFO] [1746291660.245225781] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.368905463] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.377958900] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.429388515] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.496852199] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.560946186] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.629807386] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.697596938] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.762086218] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.830085687] [showimage]: Received image #camera_frame
Received image #camera_frame
[INFO] [1746291660.893895357] [showimage]: Received image #camera_frame
Received image #camera_frame

``` 

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde