ros2/kilted_tutorial_party #277: Image Tools executables

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Image Tools](https://github.com/ros2/demos/blob/kilted/image_tools/README.md#run)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-02:
I have tested the following, and all executables in the README are present and work as expected:

## Check
* [x]  **The executables in the README are present and work as expected.**   

[image]

ahcorde, 2025-05-02:
Thank you for testing, Nice screenshot 👍

closed 2025-05-02 by ahcorde