ros2/kilted_tutorial_party #278: Improved Dynamic Discovery

tfoote, 2025-05-01:
Check the documentation for the 'Improved Dynamic Discovery' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Improved Dynamic Discovery page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Improved-Dynamic-Discovery.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-04:
- [X] **I was able to follow the documentation.**
- [X] **The documentation seemed clear to me.**
- [X] **The documentation didn't have any obvious errors.e assign this to me.**

Comments:
* Is the "Node A and B running in the different hosts" table really correct? If the nodes are on different machines then I'd expect them to not find each other if `LOCALHOST` is set. The example at the end kind of tells me otherwise though, so maybe the wording when defining `LOCALHOST` could be updated to reflect this (to me slightly surprising) behavior.

fujitatomoya, 2025-05-13:
yes that is correct, it sets the static peers, that should be discovered as long as their discovery range is not set to OFF.

closed 2025-05-13 by fujitatomoya