ros2/kilted_tutorial_party #280: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Poo-09, 2025-05-05:
Interested

Aniketkumarroy, 2025-05-10:
Tested on Windows 11 with two different versions of connext
# rti_connext_dds-5.3.1

[image]

# rti_connext_dds-7.5.0

[image]

summary:- didn't work in both, in 5.3.1 it got RMW implementation error and in 7.5.0 command exited with failure

zamantousif, 2025-05-14:
Okay, I tried this with RTI 7.5.0 and got the same issue as highlighted above.
```
(pixi_ros2_rolling) C:\dev\binary_kilted_ws>ros2 run demo_nodes_cpp talker
[ros2run]: Process exited with failure 3221226356

(pixi_ros2_rolling) C:\dev\binary_kilted_ws>ros2 run demo_nodes_cpp listener
[ros2run]: Process exited with failure 3221226356
```
`ros2 doctor` report also fails with following error, 
```
(pixi_ros2_rolling) C:\dev\binary_kilted_ws>ros2 doctor --report
C:\dev\kilted\install\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
```

**NOTE:** One issue worth mentioning is that, the `ros2 doctor --report` right after the binary setup fails with additional errors which could be fixed by installing `rosdistro` using `pip install rosdistro`. Though it did not fix the NetworkReport error as shown above.
```
(pixi_ros2_rolling) C:\dev\binary_kilted_ws>ros2 doctor --report
C:\dev\kilted\install\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\dev\kilted\install\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\dev\kilted\install\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\dev\kilted\install\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'
```

zamantousif, 2025-05-14:
@tfoote @ahcorde @mjcarroll could you please check the findings above and suggest how to proceed further?

mjcarroll, 2025-05-16:
@Aniketkumarroy and @zamantousif we are building with RTI Connext DDS 7.3.0, which is the latest long term support version.  Can you try with that rather than 7.5.0?

Aniketkumarroy, 2025-05-20:
@mjcarroll we cannot get the connext 7.3.0 version for windows, even on their website there is only 7.5.0 and 5.3.1 available. we don't know where can we get that particular version

zamantousif, 2025-05-20:
@Aniketkumarroy you need to contact the RTI guys and they will send you a link to download and test out other versions e.g. 7.3.0. I got something specific for myself on my company name. I need to find some time to try out these tickets though. Hope it helps!

zamantousif, 2025-05-20:
@mjcarroll @ahcorde I think there is enough information to close this ticket now