ros2/kilted_tutorial_party #281: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

slowrunner, 2025-05-02:
Getting same error as https://github.com/ros2/kilted_tutorial_party/issues/283

ubuntu@kilteddave:~$ ros2 run demo_nodes_cpp talker
[ERROR] [1746201022.263992080] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'librmw_connextdds.so' due to dlopen error: librmw_connextdds.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:96, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:143

 rmw_connextdds installation succeeds on Raspberry Pi / Ubuntu 24.04 but the installation does not install a working rmw_connextdds.

slowrunner, 2025-05-02:
This is the log from the rmw_connextdds install

[rmw_connextdds_install_log.txt](https://github.com/user-attachments/files/20015536/rmw_connextdds_install_log.txt)

slowrunner, 2025-05-02:
Issue has been filed with rmw_connextdds

https://github.com/ros2/rmw_connextdds/issues/190

slowrunner, 2025-05-06:
### Checks
* [x] Install works

Package was updated, and installation now works as described.
```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ dpkg -l | grep connext
ii  ros-kilted-rmw-connextdds                         1.1.0-3noble.20250502.200122             arm64        A ROS 2 RMW implementation built with RTI Connext DDS Professional.
ii  ros-kilted-rmw-connextdds-common                  1.1.0-3noble.20250502.195551             arm64        Common source for RMW implementations built with RTI Connext DDS Professional and RTI Connext DDS Micro.
ii  ros-kilted-rti-connext-dds-cmake-module           1.1.0-3noble.20250502.194515             arm64        Helper module to provide access to RTI products like Connext DDS Professional
ii  rti-connext-dds-7.3.0-ros                         7.3.0.0-5-4bullseye                      arm64        RTI Connext 7.3.0 for ROS 2
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

ubuntu@kilteddave:~/KiltedDave/ros2ws$ printenv | grep RMW
ubuntu@kilteddave:~/KiltedDave/ros2ws$ export RMW_IMPLEMENTATION=rmw_connextdds
ubuntu@kilteddave:~/KiltedDave/ros2ws$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_connextdds
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 run demo_nodes_cpp listener
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746533048.776046288] [listener]: I heard: [Hello World: 1]
[INFO] [1746533049.775609211] [listener]: I heard: [Hello World: 2]
[INFO] [1746533050.775526095] [listener]: I heard: [Hello World: 3]
[INFO] [1746533051.775485832] [listener]: I heard: [Hello World: 4]
[INFO] [1746533052.775629642] [listener]: I heard: [Hello World: 5]
[INFO] [1746533053.775582972] [listener]: I heard: [Hello World: 6]
[INFO] [1746533054.775696431] [listener]: I heard: [Hello World: 7]
^C[INFO] [1746533060.875826576] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746533060.876635145] [rclcpp]: signal_handler(signum=2)

```

Yadunund, 2025-05-06:
Great, thanks for patiently testing!

closed 2025-05-06 by Yadunund