ros2/kilted_tutorial_party #282: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Connext
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

fujitatomoya, 2025-05-02:
The installation succeeds, but unable to run the demonstration.

```console
root@tomoyafujita:~# RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
[ERROR] [1746161345.127552970] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'librmw_connextdds.so' due to dlopen error: librmw_connextdds.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:96, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:143
```

i believe this is because `ros-kilted-rmw-connextdds` does not include `librmw_connextdds.so`.

```console
root@tomoyafujita:~# dpkg -L ros-kilted-rmw-connextdds
/.
/opt
/opt/ros
/opt/ros/kilted
/opt/ros/kilted/share
/opt/ros/kilted/share/ament_index
/opt/ros/kilted/share/ament_index/resource_index
/opt/ros/kilted/share/ament_index/resource_index/package_run_dependencies
/opt/ros/kilted/share/ament_index/resource_index/package_run_dependencies/rmw_connextdds
/opt/ros/kilted/share/ament_index/resource_index/packages
/opt/ros/kilted/share/ament_index/resource_index/packages/rmw_connextdds
/opt/ros/kilted/share/ament_index/resource_index/parent_prefix_path
/opt/ros/kilted/share/ament_index/resource_index/parent_prefix_path/rmw_connextdds
/opt/ros/kilted/share/rmw_connextdds
/opt/ros/kilted/share/rmw_connextdds/cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/rmw_connextddsConfig-version.cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/rmw_connextddsConfig.cmake
/opt/ros/kilted/share/rmw_connextdds/environment
/opt/ros/kilted/share/rmw_connextdds/environment/ament_prefix_path.dsv
/opt/ros/kilted/share/rmw_connextdds/environment/ament_prefix_path.sh
/opt/ros/kilted/share/rmw_connextdds/environment/path.dsv
/opt/ros/kilted/share/rmw_connextdds/environment/path.sh
/opt/ros/kilted/share/rmw_connextdds/local_setup.bash
/opt/ros/kilted/share/rmw_connextdds/local_setup.dsv
/opt/ros/kilted/share/rmw_connextdds/local_setup.sh
/opt/ros/kilted/share/rmw_connextdds/local_setup.zsh
/opt/ros/kilted/share/rmw_connextdds/package.dsv
/opt/ros/kilted/share/rmw_connextdds/package.xml
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ros-kilted-rmw-connextdds
/usr/share/doc/ros-kilted-rmw-connextdds/changelog.Debian.gz
/usr/share/doc/ros-kilted-rmw-connextdds/copyright
```

fujitatomoya, 2025-05-02:
@fgallegosalido @lobolanja this is blocking issue to use rmw_connextdds with kilted release. can you check this issue?

fgallegosalido, 2025-05-02:
Looking into it. This is also related to https://github.com/ros2/ros2_documentation/issues/5416. Summoning @nuclearsandwich and @christophebedard for some help.

As you said @fujitatomoya, the issue is that the library `librmw_connextdds.so` is not being downloaded as part of the package `ros-kilted-rmw-connextdds` package. This happens in both Rolling and Kilted. Jazzy works fina as it used to.

fgallegosalido, 2025-05-02:
@fujitatomoya I need more info about how the package `ros-kilted-rmw-connextdds` is generated (build commands, scripts run, logs...). I can't really find any in Jenkins plan that I can correlate with the deb package. @nuclearsandwich @christophebedard

christophebedard, 2025-05-02:
I think this should be it: https://build.ros2.org/job/Kbin_uN64__rmw_connextdds__ubuntu_noble_amd64__binary/2/console

christophebedard, 2025-05-02:
Compared to the last Kbin (and even Rbin) jobs, the last Jbin job definitely does something with `librmw_connextdds.so`: https://build.ros2.org/job/Jbin_uN64__rmw_connextdds__ubuntu_noble_amd64__binary/21/consoleFull

You could try to generate the deb yourself and see what happens: https://docs.ros.org/en/rolling/How-To-Guides/Building-a-Custom-Deb-Package.html

fgallegosalido, 2025-05-02:
@christophebedard I'll try that. In the meanwhile, I updated my other comment (https://github.com/ros2/kilted_tutorial_party/issues/282#issuecomment-2847401990) with more information, just in case it may help other figure out what the problem is

fgallegosalido, 2025-05-05:
I have just tried again with the latest binaries, and it works. The new binaries ([https://build.ros2.org/job/Kbin_uN64__rmw_connextdds_common__ubuntu_noble_amd64__binary/4/](rmw_connextdds_common)) seem to work just fine. My assumption is that thew old binaries where using an older version of the buildfarm, and therefore not compiling correctly. The package `ros-kilted-rmw-connextdds` now contains the libraries:
```bash
$ dpkg -L ros-kilted-rmw-connextdds
/.
/opt
/opt/ros
/opt/ros/kilted
/opt/ros/kilted/lib
/opt/ros/kilted/lib/librmw_connextdds.so
/opt/ros/kilted/share
/opt/ros/kilted/share/ament_index
/opt/ros/kilted/share/ament_index/resource_index
/opt/ros/kilted/share/ament_index/resource_index/package_run_dependencies
/opt/ros/kilted/share/ament_index/resource_index/package_run_dependencies/rmw_connextdds
/opt/ros/kilted/share/ament_index/resource_index/packages
/opt/ros/kilted/share/ament_index/resource_index/packages/rmw_connextdds
/opt/ros/kilted/share/ament_index/resource_index/parent_prefix_path
/opt/ros/kilted/share/ament_index/resource_index/parent_prefix_path/rmw_connextdds
/opt/ros/kilted/share/ament_index/resource_index/rmw_output_prefixes
/opt/ros/kilted/share/ament_index/resource_index/rmw_output_prefixes/rmw_connextdds
/opt/ros/kilted/share/ament_index/resource_index/rmw_typesupport
/opt/ros/kilted/share/ament_index/resource_index/rmw_typesupport/rmw_connextdds
/opt/ros/kilted/share/ament_index/resource_index/rmw_typesupport_c
/opt/ros/kilted/share/ament_index/resource_index/rmw_typesupport_c/rmw_connextdds
/opt/ros/kilted/share/ament_index/resource_index/rmw_typesupport_cpp
/opt/ros/kilted/share/ament_index/resource_index/rmw_typesupport_cpp/rmw_connextdds
/opt/ros/kilted/share/rmw_connextdds
/opt/ros/kilted/share/rmw_connextdds/cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/ament_cmake_export_dependencies-extras.cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/ament_cmake_export_libraries-extras.cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/ament_cmake_export_targets-extras.cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/rmw_connextdds-extras.cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/rmw_connextddsConfig-version.cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/rmw_connextddsConfig.cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/rmw_connextddsExport-none.cmake
/opt/ros/kilted/share/rmw_connextdds/cmake/rmw_connextddsExport.cmake
/opt/ros/kilted/share/rmw_connextdds/environment
/opt/ros/kilted/share/rmw_connextdds/environment/ament_prefix_path.dsv
/opt/ros/kilted/share/rmw_connextdds/environment/ament_prefix_path.sh
/opt/ros/kilted/share/rmw_connextdds/environment/library_path.dsv
/opt/ros/kilted/share/rmw_connextdds/environment/library_path.sh
/opt/ros/kilted/share/rmw_connextdds/environment/path.dsv
/opt/ros/kilted/share/rmw_connextdds/environment/path.sh
/opt/ros/kilted/share/rmw_connextdds/local_setup.bash
/opt/ros/kilted/share/rmw_connextdds/local_setup.dsv
/opt/ros/kilted/share/rmw_connextdds/local_setup.sh
/opt/ros/kilted/share/rmw_connextdds/local_setup.zsh
/opt/ros/kilted/share/rmw_connextdds/package.dsv
/opt/ros/kilted/share/rmw_connextdds/package.xml
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ros-kilted-rmw-connextdds
/usr/share/doc/ros-kilted-rmw-connextdds/changelog.Debian.gz
/usr/share/doc/ros-kilted-rmw-connextdds/copyright
```

I think this issue can be closed along with https://github.com/ros2/rmw_connextdds/issues/190

fujitatomoya, 2025-05-05:
latest binary works without any problems.

```console
root@tomoyafujita:~# source /opt/ros/kilted/setup.bash
root@tomoyafujita:~# RMW_IMPLEMENTATION=rmw_connextdds ros2 run demo_nodes_cpp talker
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746458916.019775906] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746458917.019745979] [talker]: Publishing: 'Hello World: 2'
^C[INFO] [1746458917.240010324] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746458917.240146307] [rclcpp]: signal_handler(signum=2)
```

closed 2025-05-05 by fujitatomoya