ros2/kilted_tutorial_party #283: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Connext
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

boby2965, 2025-05-02:
Hi, I have worked on this issue.

The installation succeeds, but I am unable to run the test.

$ sudo apt update && sudo apt install -q -y rti-connext-dds-7.3.0-ros
[sudo] password for chaitanya: 
Hit:1 http://ports.ubuntu.com/ubuntu-ports noble InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports noble-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports noble-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports noble-security InRelease
Hit:5 http://packages.ros.org/ros2-testing/ubuntu noble InRelease
Reading package lists... Done                     
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists...
Building dependency tree...
Reading state information...
The following NEW packages will be installed:
  rti-connext-dds-7.3.0-ros
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 74.6 MB of archives.
After this operation, 246 MB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2-testing/ubuntu noble/main arm64 rti-connext-dds-7.3.0-ros arm64 7.3.0.0-5-4bullseye [74.6 MB]
Fetched 74.6 MB in 30s (2,504 kB/s)
Preconfiguring packages ...
rti-connext-dds-7.3.0-ros: RTI license agreement for non-commercial use accepted for all users of this system.
Selecting previously unselected package rti-connext-dds-7.3.0-ros.
(Reading database ... 224259 files and directories currently installed.)
Preparing to unpack .../rti-connext-dds-7.3.0-ros_7.3.0.0-5-4bullseye_arm64.deb ...
rti-connext-dds-7.3.0-ros: RTI license agreement for non-commercial use accepted for all users of this sytem.
Unpacking rti-connext-dds-7.3.0-ros (7.3.0.0-5-4bullseye) ...
Setting up rti-connext-dds-7.3.0-ros (7.3.0.0-5-4bullseye) ...
update-alternatives: using /usr/bin/rtienv-7.3.0 to provide /usr/bin/rtienv (rtienv) in auto mode

 
$ sudo apt update && sudo apt install -q -y ros-kilted-rmw-connextdds
Hit:1 http://packages.ros.org/ros2-testing/ubuntu noble InRelease         
Hit:2 http://ports.ubuntu.com/ubuntu-ports noble InRelease                
Hit:3 http://ports.ubuntu.com/ubuntu-ports noble-updates InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports noble-backports InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports noble-security InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  ros-kilted-rmw-connextdds-common ros-kilted-rti-connext-dds-cmake-module
The following NEW packages will be installed:
  ros-kilted-rmw-connextdds ros-kilted-rmw-connextdds-common
  ros-kilted-rti-connext-dds-cmake-module
0 upgraded, 3 newly installed, 0 to remove and 1 not upgraded.
Need to get 50.1 kB of archives.
After this operation, 326 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2-testing/ubuntu noble/main arm64 ros-kilted-rti-connext-dds-cmake-module arm64 1.1.0-2noble.20250430.210214 [11.3 kB]
Get:2 http://packages.ros.org/ros2-testing/ubuntu noble/main arm64 ros-kilted-rmw-connextdds-common arm64 1.1.0-2noble.20250501.002125 [31.6 kB]
Get:3 http://packages.ros.org/ros2-testing/ubuntu noble/main arm64 ros-kilted-rmw-connextdds arm64 1.1.0-2noble.20250501.003344 [7,180 B]
Fetched 50.1 kB in 1s (43.8 kB/s)
Selecting previously unselected package ros-kilted-rti-connext-dds-cmake-module.
(Reading database ... 225768 files and directories currently installed.)
Preparing to unpack .../ros-kilted-rti-connext-dds-cmake-module_1.1.0-2noble.20250430.210214_arm64.deb ...
Unpacking ros-kilted-rti-connext-dds-cmake-module (1.1.0-2noble.20250430.210214) ...
Selecting previously unselected package ros-kilted-rmw-connextdds-common.
Preparing to unpack .../ros-kilted-rmw-connextdds-common_1.1.0-2noble.20250501.002125_arm64.deb ...
Unpacking ros-kilted-rmw-connextdds-common (1.1.0-2noble.20250501.002125) ...
Selecting previously unselected package ros-kilted-rmw-connextdds.
Preparing to unpack .../ros-kilted-rmw-connextdds_1.1.0-2noble.20250501.003344_arm64.deb ...
Unpacking ros-kilted-rmw-connextdds (1.1.0-2noble.20250501.003344) ...
Setting up ros-kilted-rti-connext-dds-cmake-module (1.1.0-2noble.20250430.210214) ...
Setting up ros-kilted-rmw-connextdds-common (1.1.0-2noble.20250501.002125) ...
Setting up ros-kilted-rmw-connextdds (1.1.0-2noble.20250501.003344) ...

$ export RMW_IMPLEMENTATION=rmw_connextdds

$ echo $RMW_IMPLEMENTATION
rmw_connextdds

$ ros2 run demo_nodes_cpp talker
ros2: command not found

boby2965, 2025-05-02:
My Setup

$ dpkg --print-architecture
arm64
$ lsb_release -d
No LSB modules are available.
Description:	Ubuntu 24.04.2 LTS
$ sudo echo $RMW_IMPLEMENTATION
rmw_connextdds

boby2965, 2025-05-02:
after setting up the  environment, got this error

$source /opt/ros/kilted/setup.bash

$ ros2 run demo_nodes_cpp talker
[ERROR] [1746195688.487063958] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'librmw_connextdds.so' due to dlopen error: librmw_connextdds.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:96, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:143

ahcorde, 2025-05-02:
This message means that you didn't source the `setup.bash` file
```
ros2: command not found
```

if this is required or not included in the instructions, could you update the documentation ?

boby2965, 2025-05-02:
It is updated in the documentation.

slowrunner, 2025-05-02:
@ahcorde 

Same issue with https://github.com/ros2/kilted_tutorial_party/issues/281

ubuntu@kilteddave:~$ ros2 run demo_nodes_cpp talker
[ERROR] [1746201608.088427290] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'librmw_connextdds.so' due to dlopen error: librmw_connextdds.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:96, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:143

THIS IS NOT FAILED TO RUN SETUP.BASH - IT IS rmw_connextdds installation succeeds on Raspberry Pi / Ubuntu 24.04 but the installation does not install a working rmw_connextdds.

slowrunner, 2025-05-02:
Issue has been filed with rmw_connextdds

https://github.com/ros2/rmw_connextdds/issues/190

slowrunner, 2025-05-05:
With the latest update - rmw_connextdds is working on arm64

```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ export RMW_IMPLEMENTATION=rmw_connextdds
ubuntu@kilteddave:~/KiltedDave/ros2ws$ ros2 run demo_nodes_cpp listener
RTI Connext DDS Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00
Please contact support@rti.com with any questions or comments.
[INFO] [1746445247.316045391] [listener]: I heard: [Hello World: 0]
[INFO] [1746445248.223488148] [listener]: I heard: [Hello World: 1]
[INFO] [1746445249.224182061] [listener]: I heard: [Hello World: 2]
[INFO] [1746445250.224070586] [listener]: I heard: [Hello World: 3]
[INFO] [1746445251.224210609] [listener]: I heard: [Hello World: 4]
^C[INFO] [1746445254.307434904] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt
[INFO] [1746445254.308240216] [rclcpp]: signal_handler(signum=2)
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

```

The package description states "bullseye" but it is working on Ubuntu 24.04 "debian bookworm"

```
ubuntu@kilteddave:~/KiltedDave/ros2ws$ dpkg -l | grep connext
ii  ros-kilted-rmw-connextdds                         1.1.0-3noble.20250502.200122            arm64        A ROS 2 RMW implementation built with RTI Connext DDS Professional.
ii  ros-kilted-rmw-connextdds-common                  1.1.0-3noble.20250502.195551            arm64        Common source for RMW implementations built with RTI Connext DDS Professional and RTI Connext DDS Micro.
ii  ros-kilted-rti-connext-dds-cmake-module           1.1.0-3noble.20250502.194515            arm64        Helper module to provide access to RTI products like Connext DDS Professional
ii  rti-connext-dds-7.3.0-ros                         7.3.0.0-5-4bullseye                     arm64        RTI Connext 7.3.0 for ROS 2
ubuntu@kilteddave:~/KiltedDave/ros2ws$ 

```

> ## Setup
> * Rmw: Connext
> * BuildType: Debian
> * Chip: Arm64
> * Os: Noble
> 
> ## Links
> * [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)
> 
 ## Checks
 * [x]  **Install works**

Agree this ticket can be closed

fgallegosalido, 2025-05-06:
Opened issue to fix the documentation (https://github.com/ros2/ros2_documentation/issues/5557). This issue can be closed

closed 2025-05-09 by mjcarroll