ros2/kilted_tutorial_party #284: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Connext
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Aniketkumarroy, 2025-05-08:
hey I will be testig this

Aniketkumarroy, 2025-05-10:
for installation, there is no specific version mentioned for rti connext dds, so I used first 5.3.1 and ten the latest 7.5.0. it is not working in both as mentioned here #280 
if there is any specific version I should try with please tell me

zamantousif, 2025-05-14:
I tried installing latest available RTI 7.5.0 with an evaluation license. The source installation of ros2 workspace and `SET RMW_IMPLEMENTATION=rmw_connextdds` does not work due to error with loading shared library (rmw_connextdds dll) on Windows 10.
However, I was able to build an overlay workspace with rmw_connextdds following instructions from their github page and then 
 after sourcing the install script, some ros2cli commands seem to work. I have shared my findings also in another ticket https://github.com/ros2/kilted_tutorial_party/issues/389.

I think as far as the installation instructions are concerned, the RTI 7.5.0 with source install does not work out of the box. The overlay workspace once built and sourced, works but with limitations.

zamantousif, 2025-05-14:
error with source installation (setting the RMW_IMPLEMENTATION flag and trying to run ros2cli out of the box)
[image]

I followed the instructions under the [Quick Start guide on the rmw_connextdds github page](https://github.com/ros2/rmw_connextdds?tab=readme-ov-file#quick-start).

**NOTE:** The instructions require to be run as an administrator and the use of VS Command Prompt so I was running them as administrator in a `x64 Native Tools Command Prompt for VS 2019` shell. Failing to run in VS cmd prompt would throw the error `VisualStudioVersion is not set, please run within a Visual Studio Command Prompt`. After this, in the same shell I was able to build the overlay workspace with the rmw_connextdds packages and run the ros2 cli commands.

In the below screenshot, you see that after sourcing the built overlay workspace (containing the rmw_connextdds packages), simple ros2cli commands work which did not seem to be working out of the box as shown earlier.
`ros2 --help`
[image]

`ros2 node list` and other commands,
[image]

`ros2 run demo_nodes_cpp talker` still fails though
[image]

@ahcorde Is there enough information to close this ticket? Do you feel I should add issues to rmw_connextdds for the above mentioned failures?

mjcarroll, 2025-05-16:
@Aniketkumarroy and @zamantousif same here, can you try with connext 7.3.0?

Aniketkumarroy, 2025-05-20:
@mjcarroll same here, can't get the 7.3.0 version

zamantousif, 2025-05-20:
Tried with RTI 7.3.0, it works

Screenshot showing talker and listener nodes running fine from the source workspace
[image]

**NOTE:** I had to compile rmw_connextdds in an overlay workspace inside a Visual Studio cmd prompt and then source its install\setup.bat script in other command prompt terminals for this to work

[image]

@mjcarroll @ahcorde Is this enough to close the ticket?