ros2/kilted_tutorial_party #287: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Cyclone
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

slowrunner, 2025-05-02:
Checks

Yes: Install works

Installed ROS 2 Kilted on Raspberry Pi 4 arm64 Ubuntu 24.04.02 and then installed cyclonedds - both successfully completed.

T1: 
```
ubuntu@kilteddave:~$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ubuntu@kilteddave:~$ ros2 run demo_nodes_cpp listener
[INFO] [1746218351.727180013] [listener]: I heard: [Hello World: 1]
[INFO] [1746218352.726652752] [listener]: I heard: [Hello World: 2]
```

T2:
```
ubuntu@kilteddave:~$ printenv|grep RMW
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ubuntu@kilteddave:~$ ros2 run demo_nodes_cpp talker
[INFO] [1746218351.725864483] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746218352.725806720] [talker]: Publishing: 'Hello World: 2'
```

nobso, 2025-05-12:
Everything seems to be working as expected in the following env,
```
RMW Implementation: CYCLONE
BuildType: Binary
OS: Ubuntu Noble
Chip: Arm64

```

Followed the installation instruction from https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html

ROS Distro: [ros2-kilted-beta2-20250507-linux-noble-arm64.tar.bz2](https://github.com/ros2/ros2/releases/download/release-kilted-beta2-20250507/ros2-kilted-beta2-20250507-linux-noble-arm64.tar.bz2)

```
sudo apt install ros-kilted-rmw-cyclonedds-cpp
```

[image]

nobso, 2025-05-12:
Hi @ahcorde, Am I missing anything in my testing?

ahcorde, 2025-05-12:
Thank you for testing

closed 2025-05-12 by ahcorde