ros2/kilted_tutorial_party #288: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Cyclone
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

fujitatomoya, 2025-05-02:
I will take care of this.

fujitatomoya, 2025-05-02:
test log.

```console
root@tomoyafujita:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@tomoyafujita:~# ros2 run demo_nodes_cpp talker
[INFO] [1746156117.929599877] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746156118.929594552] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746156119.929563766] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746156120.929518684] [talker]: Publishing: 'Hello World: 4'
[INFO] [1746156121.929517737] [talker]: Publishing: 'Hello World: 5'
[INFO] [1746156122.929564807] [talker]: Publishing: 'Hello World: 6'
[INFO] [1746156123.929561202] [talker]: Publishing: 'Hello World: 7'
[INFO] [1746156124.929499856] [talker]: Publishing: 'Hello World: 8'
[INFO] [1746156125.929516750] [talker]: Publishing: 'Hello World: 9'
[INFO] [1746156126.929515470] [talker]: Publishing: 'Hello World: 10'
[INFO] [1746156127.929579379] [talker]: Publishing: 'Hello World: 11'
[INFO] [1746156128.929583335] [talker]: Publishing: 'Hello World: 12'
[INFO] [1746156129.929518986] [talker]: Publishing: 'Hello World: 13'

root@tomoyafujita:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@tomoyafujita:~# source /opt/ros/kilted/setup.bash
root@tomoyafujita:~# ros2 run demo_nodes_py listener
[INFO] [1746156121.956464068] [listener]: I heard: [Hello World: 5]
[INFO] [1746156122.931537321] [listener]: I heard: [Hello World: 6]
[INFO] [1746156123.931481400] [listener]: I heard: [Hello World: 7]
[INFO] [1746156124.931310613] [listener]: I heard: [Hello World: 8]
[INFO] [1746156125.931431251] [listener]: I heard: [Hello World: 9]
[INFO] [1746156126.931480182] [listener]: I heard: [Hello World: 10]
[INFO] [1746156127.931511935] [listener]: I heard: [Hello World: 11]
[INFO] [1746156128.931561318] [listener]: I heard: [Hello World: 12]
[INFO] [1746156129.931527585] [listener]: I heard: [Hello World: 13]

```

installation and demo talker/listener works without any problem.

minor suggestion here is to add `source /opt/ros/kilted/setup.bash` before `ros2 run demo_nodes_cpp talker` in https://docs.ros.org/en/kilted/Installation/RMW-Implementations/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html#run-the-talker-and-listener

closed 2025-05-02 by fujitatomoya