ros2/kilted_tutorial_party #291: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Hi, similar to https://github.com/ros2/kilted_tutorial_party/issues/293, I got an error with fastdds.
```
root@cluttered:~/ros2_kilted# rosdep install --from-paths ~/ros2_kilted/ros2-linux/share --ignore-src -y --skip-keys "cyclonedds fastcdr fastrtps iceoryx_binding_c rmw_connextdds rti-connext-dds-7.3.0 urdfdom_headers" --simulate
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure ROS_DISTRO environment variable is set, or use --rosdistro option to specify the distro, e.g. --rosdistro indigo):
rmw_fastrtps_shared_cpp: Cannot locate rosdep definition for [fastdds]
rmw_fastrtps_cpp: Cannot locate rosdep definition for [fastdds]
rmw_fastrtps_dynamic_cpp: Cannot locate rosdep definition for [fastdds]
rosidl_dynamic_typesupport_fastrtps: Cannot locate rosdep definition for [fastdds]
```

But I am not sure setting `$ROS_DISTRO` to `kilted` is the correct way according to the docs since those have not been done in humble or jazzy binary installation instructions either.

I think this is due to a previous rolling sync where `fastrtps` was renamed to `fastdds` (https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2025-03-06/42416)

I added `fastdds` to `--skip-keys ` and it worked fine. Here is the output

[image]

suchetanrs, 2025-05-02:
Some more info about my setup

Docker running on `ubuntu:24.04` image.

```
root@cluttered:~# lscpu
Architecture:             x86_64
  CPU op-mode(s):         32-bit, 64-bit
  Address sizes:          39 bits physical, 48 bits virtual
  Byte Order:             Little Endian
CPU(s):                   16
```

fujitatomoya, 2025-05-14:
@suchetanrs thanks for testing this. the issue related to `fastdds` rename, that has been addressed. i will go ahead to close this issue.

closed 2025-05-14 by fujitatomoya