ros2/kilted_tutorial_party #292: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

david-dorf, 2025-05-01:
Interested

seekbot, 2025-05-02:
Hi, not to step on anyone’s toes, but I can also confirm that Fast DDS works with the Windows binary on my [machine](https://github.com/ros2/kilted_tutorial_party/issues/286#issuecomment-2846491527). I believe @david-dorf was testing on [Windows 11](https://github.com/ros2/kilted_tutorial_party/issues/299). So it looks like the Windows binary gets a solid double-pass :)

[image]

pradyum, 2025-05-02:
In fresh installation using binary method, an error is produced while using ```ros2 doctor -r```:

```
C:\pixi_ws>ros2 doctor
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 87: UserWarning: ERROR: No loopback IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\network.py: 93: UserWarning: No multicast IP address is found.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 107: UserWarning: Check entry point PackageCheck fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 107: UserWarning: Check entry point PlatformCheck fails to load: No module named 'rosdistro'
```

It misses to install ```rosdistro``` python module. When I installed ```rosdistro``` python module manually the ```ros2 doctor -r``` was working fine

```
C:\pixi_ws>ros2 doctor -r
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.

   PACKAGE VERSIONS
actionlib_msgs                            : latest=5.5.0, local=5.5.0
action_msgs                               : latest=2.3.0, local=2.3.0
action_tutorials_cpp                      : latest=0.36.0, local=0.36.0
action_tutorials_py                       : latest=0.36.0, local=0.36.0
ament_clang_format                        : latest=0.19.2, local=0.19.2
ament_clang_tidy                          : latest=0.19.2, local=0.19.2
ament_cmake                               : latest=2.7.3, local=2.7.3
ament_cmake_auto                          : latest=2.7.3, local=2.7.3
ament_cmake_clang_format                  : latest=0.19.2, local=0.19.2
ament_cmake_clang_tidy                    : latest=0.19.2, local=0.19.2
ament_cmake_copyright                     : latest=0.19.2, local=0.19.2
ament_cmake_core                          : latest=2.7.3, local=2.7.3
ament_cmake_cppcheck                      : latest=0.19.2, local=0.19.2
ament_cmake_cpplint                       : latest=0.19.2, local=0.19.2
ament_cmake_export_definitions            : latest=2.7.3, local=2.7.3
ament_cmake_export_dependencies           : latest=2.7.3, local=2.7.3
ament_cmake_export_include_directories    : latest=2.7.3, local=2.7.3
ament_cmake_export_interfaces             : latest=2.7.3, local=2.7.3
ament_cmake_export_libraries              : latest=2.7.3, local=2.7.3
ament_cmake_export_link_flags             : latest=2.7.3, local=2.7.3
ament_cmake_export_targets                : latest=2.7.3, local=2.7.3
ament_cmake_flake8                        : latest=0.19.2, local=0.19.2
ament_cmake_gen_version_h                 : latest=2.7.3, local=2.7.3
ament_cmake_gmock                         : latest=2.7.3, local=2.7.3
ament_cmake_google_benchmark              : latest=2.7.3, local=2.7.3
ament_cmake_gtest                         : latest=2.7.3, local=2.7.3
ament_cmake_include_directories           : latest=2.7.3, local=2.7.3
ament_cmake_libraries                     : latest=2.7.3, local=2.7.3
ament_cmake_lint_cmake                    : latest=0.19.2, local=0.19.2
ament_cmake_mypy                          : latest=0.19.2, local=0.19.2
ament_cmake_pclint                        : latest=0.19.2, local=0.19.2
ament_cmake_pep257                        : latest=0.19.2, local=0.19.2
ament_cmake_pycodestyle                   : latest=0.19.2, local=0.19.2
ament_cmake_pyflakes                      : latest=0.19.2, local=0.19.2
ament_cmake_pytest                        : latest=2.7.3, local=2.7.3
ament_cmake_python                        : latest=2.7.3, local=2.7.3
ament_cmake_ros                           : latest=0.14.3, local=0.14.3
ament_cmake_ros_core                      : latest=0.14.3, local=0.14.3
ament_cmake_target_dependencies           : latest=2.7.3, local=2.7.3
ament_cmake_test                          : latest=2.7.3, local=2.7.3
ament_cmake_uncrustify                    : latest=0.19.2, local=0.19.2
ament_cmake_vendor_package                : latest=2.7.3, local=2.7.3
ament_cmake_version                       : latest=2.7.3, local=2.7.3
ament_cmake_xmllint                       : latest=0.19.2, local=0.19.2
ament_copyright                           : latest=0.19.2, local=0.19.2
ament_cppcheck                            : latest=0.19.2, local=0.19.2
ament_cpplint                             : latest=0.19.2, local=0.19.2
ament_flake8                              : latest=0.19.2, local=0.19.2
ament_index_cpp                           : latest=1.11.0, local=1.11.0
ament_index_python                        : latest=1.11.0, local=1.11.0
ament_lint                                : latest=0.19.2, local=0.19.2
ament_lint_auto                           : latest=0.19.2, local=0.19.2
ament_lint_cmake                          : latest=0.19.2, local=0.19.2
[... 301 lines cut ...]
processor        : Intel64 Family 6 Model 186 Stepping 3, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0
```

I couldn't able to get network configuration while having a warning:

```
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions
```

christophebedard, 2025-05-02:
Thanks for testing.

> It misses to install rosdistro python module. When I installed rosdistro python module manually the ros2 doctor -r was working fine

Since `ros2 doctor` isn't part of the (Windows) installation instructions, we can discuss it in another issue. I see you've opened https://github.com/ros2/ros2/pull/1683.

closed 2025-05-02 by christophebedard