ros2/kilted_tutorial_party #293: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Fastdds
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-01:
Hi, can you assign this ticket to me?

danielcranston, 2025-05-01:
My setup:
```
$ dpkg --print-architecture
arm64
$ lsb_release -d
No LSB modules are available.
Description:    Ubuntu 24.04.2 LTS
```

I followed the instructions in https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html.

Worked fine overall, just had one small question and a minor issue.

### Question: Which binary archive to download?
In the [Install ROS 2](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html#install-ros-2) instructions, it points to [this page](https://docs.ros.org/en/kilted/Installation/Testing.html#prerelease-binaries), which says to download the binary archive from https://ci.ros2.org/view/packaging/. However, based on [this post](https://discourse.ros.org/t/ros-2-kilted-beta-and-call-for-testing/43477) I get the impression that I should be downloading it from the Assets of the [Kilted Github pre-release](https://github.com/ros2/ros2/releases/tag/release-kilted-beta-20250430).

I downloaded both and ended up extracting the one from the Github pre-release Asset, since it finished first (ci.ros2.org download speed was very slow for some reason). I'd like to make sure this is correct.

### Small issue: rosdep install command could not find `fastdds`

As per [the instructions](https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html#install-dependencies-using-rosdep), I ran 

```
$ rosdep install --from-paths ~/ros2_kilted_binary_archive/ros2-linux/share --ignore-src -y --skip-keys "cyclonedds fastcdr fastrtps iceoryx_binding_c rmw_connextdds rti-connext-dds-7.3.0 urdfdom_headers"
```

However, I got the below ERROR, along with `fastdds` not being found as a rosdep definition:
```
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure `ROS_DISTRO` environment variable is set, or use `--rosdistro` option to specify the distro, e.g. `--rosdistro indigo`):
rosidl_dynamic_typesupport_fastrtps: Cannot locate rosdep definition for [fastdds]
rmw_fastrtps_dynamic_cpp: Cannot locate rosdep definition for [fastdds]
rmw_fastrtps_shared_cpp: Cannot locate rosdep definition for [fastdds]
rmw_fastrtps_cpp: Cannot locate rosdep definition for [fastdds]
```

Appending `--rosdistro kilted` resolves the issue, `fastdds` could beresolved and was installed.

### Verifying installation
I verified the installation by running demo talker/listener nodes in two separate terminals:

```
$ source ~/ros2_kilted_binary_archive/ros2-linux/setup.bash 
$ ros2 run demo_nodes_cpp talker
[INFO] [1746118160.171085441] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746118161.171283251] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746118162.172492211] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746118163.171266941] [talker]: Publishing: 'Hello World: 4'
...
```

```

$ source ~/ros2_kilted_binary_archive/ros2-linux/setup.bash 
$ ros2 run demo_nodes_py listener
[INFO] [1746118180.751572873] [listener]: I heard: [Hello World: 18]
[INFO] [1746118181.700118783] [listener]: I heard: [Hello World: 19]
```

danielcranston, 2025-05-01:
I can't edit the issue description, but barring the small rosdep thing above,

* [X]  **Install works**

christophebedard, 2025-05-03:
Thanks for testing!

> Question: Which binary archive to download?

I think this section in the Kilted docs shouldn't mention the prerelease binaries. Those are only Rolling, except for when the package jobs are triggered manually to generate binaries that then get uploaded as a release on GitHub. I believe it is mentioned currently because the Kilted docs were created from the Rolling docs.

So you should use the Kilted "pre-release" binaries from GitHub. I've opened a PR to remove the part that mentions the (Rolling) prerelease binaries there: https://github.com/ros2/ros2_documentation/pull/5483.

> Small issue: rosdep install command could not find fastdds

I think https://github.com/ros2/ros2_documentation/pull/5482 should fix that.

closed 2025-05-03 by christophebedard