ros2/kilted_tutorial_party #295: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Chip: Arm64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2025-05-01:
PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.11.0-24-generic-aarch64-with-glibc2.39
release          : 6.11.0-24-generic
processor        : aarch64

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

**OUTPUT**

The installation instructions are clear and the demo programs are working as expected.

christophebedard, 2025-05-02:
Thanks for testing!

closed 2025-05-02 by christophebedard