ros2/kilted_tutorial_party #296: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Fastdds
- BuildType: Source
- Chip: Amd64
- Os: Rhel 9

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-07:
Against my better judgement, I decided to give this a try.

I used [some random rhel-9 docker image](https://hub.docker.com/layers/alvistack/rhel-9/20230618.1.1/images/sha256-673d90022a41ab5010bc93397f0f67fde74ab36401386e78c3619b1ed951a9d2) I found on docker hub, but in hindsight I should have probably used https://hub.docker.com/r/redhat/ubi9.

<details>
  <summary>"sudo dnf update" - Some errors</summary>

```
# sudo dnf update
Updating Subscription Management repositories.
Unable to read consumer identity
Subscription Manager is operating in container mode.

This system is not registered with an entitlement server. You can use subscription-manager to register.

Last metadata expiration check: 0:19:46 ago on Wed 07 May 2025 10:47:35 AM UTC.
Error: 
 Problem 1: package ansible-100:8.7.0-8.20.noarch requires ansible-core >= 100:2.15.7, but none of the providers can be installed
  - cannot install the best update candidate for package ansible-100:8.0.0-1.4.noarch
  - package ansible-core-100:2.15.13-3.19.noarch is filtered out by exclude filtering
 Problem 2: package python3-coverage+toml-6.2-1.el9.x86_64 requires python3-coverage = 6.2-1.el9, but none of the providers can be installed
  - cannot install both python3-coverage-100:7.8.0-1.10.x86_64 and python3-coverage-6.2-1.el9.x86_64
  - cannot install both python3-coverage-6.2-1.el9.x86_64 and python3-coverage-100:7.8.0-1.10.x86_64
  - cannot install the best update candidate for package python3-coverage+toml-6.2-1.el9.x86_64
  - cannot install the best update candidate for package python3-coverage-6.2-1.el9.x86_64
 Problem 3: package ansible-core-100:2.15.0-2.8.noarch requires (python3.9dist(resolvelib) < 1.1 with python3.9dist(resolvelib) >= 0.5.3), but none of the providers can be installed
  - cannot install both python3-resolvelib-100:1.1.0-3.20.noarch and python3-resolvelib-100:1.0.1-1.23.noarch
  - cannot install both python3-resolvelib-0.5.4-5.el9.noarch and python3-resolvelib-100:1.1.0-3.20.noarch
  - cannot install the best update candidate for package python3-resolvelib-100:1.0.1-1.23.noarch
  - problem with installed package ansible-core-100:2.15.0-2.8.noarch
```
</details>

<details>
  <summary>"rosdep install" - Fails to install a bunch of stuff</summary>

```
# rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-7.3.0 urdfdom_headers"
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
executing command [dnf --assumeyes --setopt=strict=0 install python3-pytest-timeout python3-numpy opencv-devel gtest-devel python3-qt5-devel python3-sip-devel libXext-devel qt5-qtbase-devel libyaml-devel python3-devel python3-pydocstyle cargo clang python3-yaml google-benchmark-devel cmake3 python3-lark-parser spdlog-devel eigen3-devel python3-flake8-blind-except python3-flake8-builtins python3-flake8-class-newline python3-flake8-comprehensions python3-flake8-deprecated python3-flake8-docstrings python3-flake8-import-order python3-flake8-quotes gmock-devel python3-pydot python3-pygraphviz pybind11-devel console-bridge-devel cppcheck libsq3-devel tinyxml2-devel python3-pykdl bullet-devel qt5-qtbase qt5-qtbase-gui graphviz python3-babeltrace lttng-tools lttng-ust-devel libacl-devel libatomic python3-fastjsonschema libcurl-devel uncrustify clang-tools-extra python3-psutil assimp-devel tango-icon-theme ncurses-devel libzstd-devel lz4-devel clang-tools-extra git-clang-format libX11-devel libXaw-devel libXrandr-devel mesa-libGL-devel mesa-libGLU-devel freetype-devel glew-devel asio-devel orocos-kdl-devel lttng-tools-devel json-devel qt5-qtsvg qt5-qtsvg-devel yaml-cpp-devel python3-pillow python3-matplotlib]
Updating Subscription Management repositories.
Unable to read consumer identity

This system is not registered with an entitlement server. You can use subscription-manager to register.

Last metadata expiration check: 0:21:40 ago on Wed 07 May 2025 10:47:35 AM UTC.
No match for argument: python3-qt5-devel
No match for argument: python3-sip-devel
No match for argument: libXext-devel
No match for argument: qt5-qtbase-devel
No match for argument: python3-lark-parser
No match for argument: eigen3-devel
No match for argument: qt5-qtbase
No match for argument: qt5-qtbase-gui
No match for argument: python3-babeltrace
No match for argument: lttng-ust-devel
No match for argument: libacl-devel
No match for argument: libXaw-devel
No match for argument: libXrandr-devel
No match for argument: mesa-libGL-devel
No match for argument: mesa-libGLU-devel
No match for argument: qt5-qtsvg
No match for argument: qt5-qtsvg-devel
Dependencies resolved.

 Problem 1: conflicting requests
  - nothing provides mesa-libGLU-devel(x86-64) needed by glew-devel-2.2.0-2.el9.x86_64
  - nothing provides pkgconfig(glu) needed by glew-devel-2.2.0-2.el9.x86_64
 Problem 2: conflicting requests
  - nothing provides libqhull_r.so.7()(64bit) needed by python3-matplotlib-3.4.3-4.el9.x86_64
[... 627 lines cut ...]
# colcon build --symlink-install
...
--- stderr: iceoryx_hoofs                                                                                                   
In file included from /root/ros2_kilted/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/include/iceoryx_hoofs/internal/posix_wrapper/access_control.hpp:23,
                 from /root/ros2_kilted/src/eclipse-iceoryx/iceoryx/iceoryx_hoofs/source/posix_wrapper/access_control.cpp:18:
[... cut ...]
Failed   <<< iceoryx_hoofs [16.9s, exited with code 2]
```

</details>

Long story short, I could not get this to build in this image. I could install development tools and `vcs import` the sources, but rosdep and actual building failed.

I've never used Rhel-9 either, so I might be missing something obvious. I'll try again with the redhat official image.

---

One comment: If dnf is any thing like apt, I think the instructions should do `dnf update` before the first mention of `dnf install`. That currently is not the case.

danielcranston, 2025-05-07:
- [x] **Install works**

I ended up using `almalinux:9` because I had some issues enabling `crb` inside `redhat/ubi9`, and because it seems to be used with success here: https://github.com/ros-controls/ros2_control_ci/blob/master/ros2_rhel/Dockerfile.rhel9

Indeed it was smooth sailing after finding this image. Enabling `crb`, installing devtools, cloning the sources with `vcs`, installing binary dependencies via `rosdep install`, and building with `colcon` all worked without a hitch.

Although I guess it's some derived flavor of Rhel-9, rosdep understands it as Rhel-9, which is nice.
```
# rosdep update
...
rosdep detected OS: [almalinux] aliasing it to: [rhel]
...
Add distro "kilted"
...
```

### Setup
```bash
# lscpu | grep Architecture
Architecture:                         x86_64

# cat /etc/os-release
NAME="AlmaLinux"
VERSION="9.5 (Teal Serval)"
ID="almalinux"
ID_LIKE="rhel centos fedora"
VERSION_ID="9.5"
PLATFORM_ID="platform:el9"
PRETTY_NAME="AlmaLinux 9.5 (Teal Serval)"
ANSI_COLOR="0;34"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:almalinux:almalinux:9::baseos"
HOME_URL="https://almalinux.org/"
DOCUMENTATION_URL="https://wiki.almalinux.org/"
BUG_REPORT_URL="https://bugs.almalinux.org/"

ALMALINUX_MANTISBT_PROJECT="AlmaLinux-9"
ALMALINUX_MANTISBT_PROJECT_VERSION="9.5"
REDHAT_SUPPORT_PRODUCT="AlmaLinux"
REDHAT_SUPPORT_PRODUCT_VERSION="9.5"
SUPPORT_END=2032-06-01
```

### Checks

I followed https://docs.ros.org/en/kilted/Installation/Alternatives/RHEL-Development-Setup.html.

<details>
  <summary>Installing build tools</summary>

```
# dnf install -y   cmake   gcc-c++   git   make   patch   python3-colcon-common-extensions   python3-mypy   python3-pip   python3-pytest   python3-pytest-cov   python3-pytest-mock   python3-pytest-repeat   python3-pytest-rerunfailures   python3-pytest-runner   python3-rosdep   python3-setuptools   python3-vcstool   wget
AlmaLinux 9 - CRB                                                                                                       987 kB/s | 4.0 MB     00:04    
Extra Packages for Enterprise Linux 9 - x86_64                                                                          617 kB/s |  23 MB     00:38    
Extra Packages for Enterprise Linux 9 openh264 (From Cisco) - x86_64                                                    673  B/s | 2.5 kB     00:03    
Dependencies resolved.
========================================================================================================================================================
 Package                                                    Architecture         Version                                  Repository               Size
========================================================================================================================================================
Installing:
 cmake                                                      x86_64               3.26.5-2.el9                             appstream               8.7 M
 gcc-c++                                                    x86_64               11.5.0-5.el9_5.alma.1                    appstream                13 M
 git                                                        x86_64               2.43.5-2.el9_5                           appstream                50 k
 make                                                       x86_64               1:4.3-8.el9                              baseos                  530 k
 patch                                                      x86_64               2.7.6-16.el9                             appstream               127 k
 python3-colcon-common-extensions                           noarch               0.3.0-1.el9                              epel                     15 k
 python3-mypy                                               noarch               0.971-2.el9                              epel                    2.7 M
 python3-pip                                                noarch               21.3.1-1.el9                             appstream               1.7 M
 python3-pytest                                             noarch               6.2.2-7.el9                              appstream               473 k
 python3-pytest-cov                                         noarch               4.0.0-2.el9                              epel                     41 k
[... 1287 lines cut ...]
 polkit-libs                                          x86_64              0.117-13.el9                                     baseos                 8.3 M
 poly2tri                                             x86_64              0.0-21.20130501hg26242d0aa7b8.el9                epel                    31 k
 poly2tri-devel                                       x86_64              0.0-21.20130501hg26242d0aa7b8.el9                epel                    17 k
 poppler                                              x86_64              21.01.0-21.el9                                   appstream              1.1 M
 poppler-data                                         noarch              0.4.9-9.el9                                      appstream              1.8 M
 poppler-glib                                         x86_64              21.01.0-21.el9                                   appstream              151 k
 procps-ng                                            x86_64              3.3.17-14.el9                                    baseos                 332 k
[... cut ...]
 python3-matplotlib-data-fonts                        noarch              3.4.3-4.el9                                      epel                   2.3 M
 python3-matplotlib-tk                                x86_64              3.4.3-4.el9                                      epel                    35 k
 python3-mccabe                                       noarch              0.6.1-22.el9                                     epel                    19 k
 python3-olefile                                      noarch              0.46-15.el9                                      epel

cottsay, 2025-05-09:
> One comment: If dnf is any thing like apt, I think the instructions should do dnf update before the first mention of dnf install. That currently is not the case.

`dnf` updates metadata as necessary - there is no equivalent of `apt update`.

Using AlmaLinux is perfectly acceptable. Thanks for testing!

closed 2025-05-09 by cottsay