ros2/kilted_tutorial_party #297: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Fastdds
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Matt5sean3, 2025-05-01:
Could I get assigned to this ticket?

Matt5sean3, 2025-05-02:
A few notes so far:

- Windows 11 can work, so the doc shouldn't rule it out.
- For enabling long paths, it should be noted that Powershell needs to be open with admin privileges.
- Using --passive instead of --quiet for MSVC installation would make it less inscrutable
- The step to download Pixi ends by saying to close and re-open Powershell, then the following step contradicts this by saying to use the existing powershell session
- The downloaded pixi file at https://raw.githubusercontent.com/ros2/ros2/refs/heads/kilted/pixi.toml has the name "pixi_ros2_rolling" and is reflected as such in the terminal after running pixi shell.
- google_benchmark_vendor fails with "No CMAKE_CXX_COMPILER could be found" and gtest_vendor failed with "could not find any instance of Visual Studio" unless you restart your computer after installing Visual Studio.
- The build for fastdds failed with an Internal compiler error. The full stdout log is attached, but from reading elsewhere it's a bug that [exists in Visual Studio 2019 that is patched in Visual Studio 2022 and later with no plans to backport to VS 2019](https://developercommunity.visualstudio.com/t/fatal-error-c1001-internal-compiler-error-compiler-7/1555890).

[stdout.log](https://github.com/user-attachments/files/20007789/stdout.log)

Matt5sean3, 2025-05-02:
I went back a few steps to switch to Visual Studio 2022 by uninstalling VS2019 then using

`irm https://aka.ms/vs/17/release/vs_buildtools.exe -OutFile vs_buildtools_2022.exe`

`.\vs_buildtools_2022.exe --wait --add Microsoft.Component.MSBuild --add Microsoft.Net.Component.4.6.1.TargetingPack --add Microsoft.Net.Component.4.8.SDK --add Microsoft.VisualStudio.Component.CoreBuildTools --add Microsoft.VisualStudio.Component.Roslyn.Compiler --add Microsoft.VisualStudio.Component.TextTemplating --add Microsoft.VisualStudio.Component.VC.CLI.Support --add Microsoft.VisualStudio.Component.VC.CoreBuildTools --add Microsoft.VisualStudio.Component.VC.CoreIde --add Microsoft.VisualStudio.Component.VC.Redist.14.Latest --add Microsoft.VisualStudio.Component.VC.Tools.x86.x64 --add Microsoft.VisualStudio.Component.Windows10SDK --add Microsoft.VisualStudio.Component.Windows10SDK.19041 --add Microsoft.VisualStudio.ComponentGroup.NativeDesktop.Core --add Microsoft.VisualStudio.Workload.MSBuildTools --add Microsoft.VisualStudio.Workload.VCTools`

Restart my computer, open a new command prompt

`call "C:\Program Files (x86)\Microsoft Visual Studio\2022\BuildTools\VC\Auxiliary\Build\vcvarsall.bat" x86_amd64`

I then follow instructions up to starting compilation at which point iceoryx_hoofs fails due to a different internal compiler error. [stdout_stderr.log](https://github.com/user-attachments/files/20016597/stdout_stderr.log)

dilpreetsingh25, 2025-05-12:
using Window11 and default is Fastdds

**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.11.46
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************

C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools>cd C:\dev
C:\dev>pixi shell
(pixi_ros2_rolling) C:\dev>cd C:\dev\kilted
(pixi_ros2_rolling) C:\dev\kilted>cd C:\dev\kilted
(pixi_ros2_rolling) C:\dev\kilted>echo %RMW_IMPLEMENTATION%
%RMW_IMPLEMENTATION%

this below command works in Developer Command prompt VS 2019 that should be mentioned in documentation.
this below command is going to take 1-2hr to complete depending on configuration.

(pixi_ros2_rolling) C:\dev\kilted>colcon build --merge-install
[109.893s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'iceoryx_introspection' because it has no 'install' target
[164.908s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'tlsf' because it has no 'install' target
[165.147s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'rttest' because it has no 'install' target
[1545.070s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'rosbag2_performance_benchmarking_msgs' because it has no 'install' target
[3399.522s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'tlsf_cpp' because it has no 'install' target
[3549.343s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'pendulum_control' because it has no 'install' target
[5431.706s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'rosbag2_performance_benchmarking' because it has no 'install' target
--
--
Finished <<< interactive_markers [3min 24s]
--- stderr: test_tf2
--
--
CMake Warning (dev) at C:/dev/kilted/install/share/tf2_bullet/cmake/bullet-extras.cmake:35 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
  CMake variable BULLET_ROOT is set to:
    /lib/bullet
  For compatibility, find_package is ignoring the variable, but code in a
  .cmake module might still use it.
Call Stack (most recent call first):
  C:/dev/kilted/install/share/tf2_bullet/cmake/tf2_bulletConfig.cmake:41 (include)
  CMakeLists.txt:34 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.
--
--
Summary: 370 packages finished [1h 37min 51s]
  2 packages had stderr output: test_tf2 tf2_bullet

FIREWALL INSTRUCTION can be provided.

What you can do:
Option 1: Allow them all once
If you’re confident all executables are from trusted ROS 2 sources:
Click "Allow access" each time(100+ times) the firewall popup appears.
Next time you run the tests, these permissions are cached, and you won’t be prompted again.

Option 2: Whitelist your whole ROS 2 workspace
You can manually whitelist executables from the install/ and build/ folders in Windows Defender Firewall:
Open Windows Security > Firewall & network protection.
Click Allow an app through firewall.
Add relevant .exe files from:
install\lib\*\*.exe
build\*\*.exe (only if needed)

Option 3: Run tests with Firewall temporarily disabled (Not recommended for long term)
Disable firewall temporarily while running tests:
Control Panel > Windows Defender Firewall > Turn off firewall
Re-enable afterward.
⚠️ Security risk: Only do this offline or if you trust the network.

[... 25 lines cut ...]
---
Finished <<< rqt_bag [1min 38s]
--- stderr: ros2bag

============================== warnings summary ===============================
test/test_burst.py::test_burst
  Warning: Cannot create a symlink to latest log directory: [WinError 1314] A required privilege is not held by the client: 'C:\\Users\\Dilpreet Singh\\.ros\\log\\2025-05-12-01-19-43-115597-MSI-28716' -> 'C:\\Users\\Dilpreet Singh\\.ros\\log\\latest'
test/test_flake8.py::test_flake8
  Warning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
---
and many more WARNINGS SUMMARY ---------------------
--
--
--
============================== warnings summary ===============================
test/test_flake8.py::test_flake8
  Warning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
---
Finished <<< rqt_bag_plugins [1min 33s]
Summary: 370 packages finished [1h 31min 23s]
  89 packages had stderr output: action_tutorials_py ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_uncrustify ament_xmllint camera_info_manager_py demo_nodes_cpp demo_nodes_py domain_coordinator examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher examples_tf2_py image_tools i [... cut ...] qt_publisher rqt_py_console rqt_reconfigure rqt_service_caller rqt_shell rqt_srv rqt_topic sensor_msgs_py sros2 test_communication test_launch_ros test_quality_of_service test_rclcpp test_ros2trace test_tracetools_launch tf2_ros_py tf2_tools topic_monitor tracetools_launch tracetools_read tracetools_test tracetools_trace
  11 packages had test failures: demo_nodes_cpp image_tools intra_process_demo logging_demo rcl rclcpp rosbag2_tests sros2 test_communication test_quality_of_service test_rclcpp

dilpreetsingh25, 2025-05-12:
`(pixi_ros2_rolling) C:\dev\kilted>ros2 run demo_nodes_cpp talker
[INFO] [1746995218.660624700] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746995219.664260400] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746995220.666773700] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746995221.665218700] [talker]: Publishing: 'Hello World: 4'
[INFO] [1746995222.666628800] [talker]: Publishing: 'Hello World: 5'
[INFO] [1746995223.667526300] [talker]: Publishing: 'Hello World: 6'
[INFO] [1746995224.668415800] [talker]: Publishing: 'Hello World: 7'
[INFO] [1746995225.666865100] [talker]: Publishing: 'Hello World: 8'
[INFO] [1746995226.665684100] [talker]: Publishing: 'Hello World: 9'
[INFO] [1746995227.668167300] [talker]: Publishing: 'Hello World: 10'
[INFO] [1746995228.680219800] [talker]: Publishing: 'Hello World: 11'
[INFO] [1746995229.671422100] [talker]: Publishing: 'Hello World: 12'
[INFO] [1746995230.048540700] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\dev\kilted>ros2 run demo_nodes_cpp talker
[INFO] [1746995246.158513700] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746995247.166507100] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746995248.178556200] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746995248.983893400] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\dev\kilted>`

`(pixi_ros2_rolling) C:\dev\kilted>ros2 run demo_nodes_py listener
[INFO] [1746995218.785566500] [listener]: I heard: [Hello World: 1]
[INFO] [1746995219.692737700] [listener]: I heard: [Hello World: 2]
[INFO] [1746995220.670298700] [listener]: I heard: [Hello World: 3]
[INFO] [1746995221.666781100] [listener]: I heard: [Hello World: 4]
[INFO] [1746995222.668608200] [listener]: I heard: [Hello World: 5]
[INFO] [1746995223.668786600] [listener]: I heard: [Hello World: 6]
[INFO] [1746995224.670108200] [listener]: I heard: [Hello World: 7]
[INFO] [1746995225.668376900] [listener]: I heard: [Hello World: 8]
[INFO] [1746995226.668453100] [listener]: I heard: [Hello World: 9]
[INFO] [1746995227.669765800] [listener]: I heard: [Hello World: 10]
[INFO] [1746995228.687429700] [listener]: I heard: [Hello World: 11]
[INFO] [1746995229.672784800] [listener]: I heard: [Hello World: 12]
[INFO] [1746995246.179328900] [listener]: I heard: [Hello World: 1]
[INFO] [1746995247.172758700] [listener]: I heard: [Hello World: 2]
[INFO] [1746995248.180618700] [listener]: I heard: [Hello World: 3]
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\dev\kilted>`

## Installlation Works

zamantousif, 2025-05-14:
Good point above about the installation working in Windows 11.

Since the original instructions pointed to Windows 10, I tried them out independently on a Windows 10 machine. The source installation instructions work although there should be some note that the `colcon build --merge-install` in the ros2 workspace takes a very long time (in my case, it was taking forever so I left it overnight) as it pulls around 100-200 packages, compiles them so a small note about this waiting time in the instructions could be good. 

As far as the scope of the installation in the ticket is concerned, I was able to run the talker, listener node and the screenshot below shows that.

[image]

zamantousif, 2025-05-14:
I added a note about the long build times to the documentation [in this PR](https://github.com/ros2/ros2_documentation/pull/5620). I did not however, include the information about supporting Windows 11 yet.

zamantousif, 2025-05-14:
@ahcorde I suppose this ticket has enough information and results from both Windows 10 and Windows 11. 
Could you kindly take a look and close it please?

ahcorde, 2025-05-14:
PR is merged, thank you for testing

closed 2025-05-14 by ahcorde

zamantousif, 2025-05-14:
Thanks @ahcorde ! Could you kindly assign us to the ticket as well?