ros2/kilted_tutorial_party #298: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Testing this

suchetanrs, 2025-05-02:
I followed https://docs.ros.org/en/kilted/Installation/Alternatives/Ubuntu-Install-Binary.html

Like mentioned in https://github.com/ros2/kilted_tutorial_party/issues/291#issuecomment-2846725081

i had to add the `fastdds` to `--skip-keys`. After that, it worked

[image]

closed 2025-05-14 by fujitatomoya