ros2/kilted_tutorial_party #299: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Amd64
- Os: Windows

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

david-dorf, 2025-05-01:
Interested in this for Windows 11

david-dorf, 2025-05-01:
Binary installation works as expected on Windows 11 (AMD64). Had to do some additional steps (within `pixi shell` in Terminal):

- In each terminal `set RMW_IMPLEMENTATION=rmw_zenoh_cpp` after sourcing
- In a third terminal, `ros2 run rmw_zenoh_cpp rmw_zenohd` to enable the talker listener pair to work

Worth noting:

- It might be good to warn users that Windows will alert you when installing conda forge and running anything on the network

https://github.com/user-attachments/assets/2505d7bd-8b16-4874-ba8a-17957930d514

christophebedard, 2025-05-02:
Interesting, what does the conda forge alert look like?

david-dorf, 2025-05-02:
It shows this on my system, have to click "More Info" [not shown] to get the "Run Anyway" option to show.

[image]

david-dorf, 2025-05-02:
> Thanks for the info! I indeed think it would be worth opening a PR to mention this as a note in the Windows installation instructions, just below whatever command leads to this alert. Would you be willing to open a PR?

Sure! https://github.com/ros2/ros2_documentation/pull/5479

christophebedard, 2025-05-02:
Great, thanks for opening the PR and for validating!