ros2/kilted_tutorial_party #3: Adding A Frame Py

tfoote, 2025-05-01:
Check the documentation for the 'Adding A Frame Py' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Adding A Frame Py page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Adding-A-Frame-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Aniketkumarroy, 2025-05-03:
hey I will take this up please assign me to this

Aniketkumarroy, 2025-05-04:
ran the test and here the results
### **Adding a fixed frame**

[image]

### **following the fixed frame**

[image]

### **following a dynamic frame**

[image]

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde