ros2/kilted_tutorial_party #300: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Zenoh
- BuildType: Binary
- Chip: Arm64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

danielcranston, 2025-05-02:
Identical to https://github.com/ros2/kilted_tutorial_party/issues/293#issuecomment-2845258470, except here Zenoh should be used. Since `rmw_zenoh_cpp` is already included in the binary archive (tarball), no building from source is needed, and just exporting the environment variable is enough:

```
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
```

* [X]  **Install works**

Following https://docs.ros.org/en/kilted/Installation/RMW-Implementations/Non-DDS-Implementations/Working-with-Zenoh.html

```bash
# Terminal 1
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 run rmw_zenoh_cpp rmw_zenohd 
2025-05-02T08:11:47.527401Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: 4f5f0e3d5cdd365185ef727e73483d8f
2025-05-02T08:11:47.529019Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::215:5dff:fe17:c26f]:7447
2025-05-02T08:11:47.529040Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/10.255.255.254:7447
2025-05-02T08:11:47.529042Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/172.27.238.129:7447
Started Zenoh router with id 4f5f0e3d5cdd365185ef727e73483d8f
```

```bash
# Terminal 2
$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
$ ros2 run demo_nodes_cpp talker
[INFO] [1746173516.073599029] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746173517.073468185] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746173518.073550379] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746173519.073407710] [talker]: Publishing: 'Hello World: 4'
...
```

```bash
# Terminal 3
$ ros2 run demo_nodes_cpp listener
[INFO] [1746173524.076757106] [listener]: I heard: [Hello World: 9]
[INFO] [1746173525.075659070] [listener]: I heard: [Hello World: 10]
[INFO] [1746173526.075425924] [listener]: I heard: [Hello World: 11]
[INFO] [1746173527.076144846] [listener]: I heard: [Hello World: 12]
```

---

I think there's a typo in https://docs.ros.org/en/kilted/Installation/RMW-Implementations/Non-DDS-Implementations/Working-with-Zenoh.html#run-the-talker-and-listener though.

> Start the Zenoh router
> 
> ```
> # terminal 1
> ros2 run rmw_zenoh_cpp talker rmw_zenohd
> ```

This is a typo, and the intended command is `ros2 run rmw_zenoh_cpp rmw_zenohd`, right?

ahcorde, 2025-05-02:
Thank you for testing and openning the PR on ros2_documentation!

closed 2025-05-02 by ahcorde