ros2/kilted_tutorial_party #301: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Zenoh
- BuildType: Debian
- Chip: Amd64
- Os: Noble

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Amronos, 2025-05-02:
Hello! I would like to work on this ticket. Can it please be assigned to me?

Amronos, 2025-05-02:
Followed https://docs.ros.org/en/kilted/Installation/Ubuntu-Install-Debs.html and https://docs.ros.org/en/kilted/Installation/RMW-Implementations/Non-DDS-Implementations/Working-with-Zenoh.html.

 - [x] Install works

```console
amronos@dev-pc:~$ ros2 run rmw_zenoh_cpp rmw_zenohd
2025-05-02T14:47:07.540438Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: fb31cd6ed6f8d440d043e68df607059b
2025-05-02T14:47:07.541764Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2401:4900:1c66:f8bf:6f74:211d:e60a:ad61]:7447
2025-05-02T14:47:07.541807Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2401:4900:1c66:f8bf:85:b5d9:b19b:b04c]:7447
2025-05-02T14:47:07.541817Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::4041:301d:80a2:7e90]:7447
2025-05-02T14:47:07.541830Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.1.5:7447
Started Zenoh router with id fb31cd6ed6f8d440d043e68df607059b

amronos@dev-pc:~$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
amronos@dev-pc:~$ ros2 run demo_nodes_cpp talker
[INFO] [1746197266.094027120] [talker]: Publishing: 'Hello World: 1'
[INFO] [1746197267.093947115] [talker]: Publishing: 'Hello World: 2'
[INFO] [1746197268.093986931] [talker]: Publishing: 'Hello World: 3'
[INFO] [1746197269.094435276] [talker]: Publishing: 'Hello World: 4'
[INFO] [1746197270.093940724] [talker]: Publishing: 'Hello World: 5'
[INFO] [1746197271.093941192] [talker]: Publishing: 'Hello World: 6'
[INFO] [1746197272.094105061] [talker]: Publishing: 'Hello World: 7'
[INFO] [1746197273.094509367] [talker]: Publishing: 'Hello World: 8'
[INFO] [1746197274.094013395] [talker]: Publishing: 'Hello World: 9'
[INFO] [1746197275.094387706] [talker]: Publishing: 'Hello World: 10'
[INFO] [1746197276.093973691] [talker]: Publishing: 'Hello World: 11'
[INFO] [1746197277.093980357] [talker]: Publishing: 'Hello World: 12'
[INFO] [1746197278.093945241] [talker]: Publishing: 'Hello World: 13'
[INFO] [1746197279.093944566] [talker]: Publishing: 'Hello World: 14'
[INFO] [1746197280.094342250] [talker]: Publishing: 'Hello World: 15'
[INFO] [1746197281.094279840] [talker]: Publishing: 'Hello World: 16'
[INFO] [1746197282.093946190] [talker]: Publishing: 'Hello World: 17'
[INFO] [1746197283.094011439] [talker]: Publishing: 'Hello World: 18'
[INFO] [1746197284.093944200] [talker]: Publishing: 'Hello World: 19'
[INFO] [1746197285.094011484] [talker]: Publishing: 'Hello World: 20'
[INFO] [1746197286.094021158] [talker]: Publishing: 'Hello World: 21'
[INFO] [1746197287.094000732] [talker]: Publishing: 'Hello World: 22'
[INFO] [1746197288.094548632] [talker]: Publishing: 'Hello World: 23'
[INFO] [1746197289.093951139] [talker]: Publishing: 'Hello World: 24'
[INFO] [1746197290.093938593] [talker]: Publishing: 'Hello World: 25'
[INFO] [1746197291.093953515] [talker]: Publishing: 'Hello World: 26'
[INFO] [1746197292.093961100] [talker]: Publishing: 'Hello World: 27'
[INFO] [1746197293.093943092] [talker]: Publishing: 'Hello World: 28'
[INFO] [1746197294.093949204] [talker]: Publishing: 'Hello World: 29'
[INFO] [1746197295.093966624] [talker]: Publishing: 'Hello World: 30'

amronos@dev-pc:~$ export RMW_IMPLEMENTATION=rmw_zenoh_cpp
amronos@dev-pc:~$ ros2 run demo_nodes_cpp listener
[INFO] [1746197284.094812322] [listener]: I heard: [Hello World: 19]
[INFO] [1746197285.094624490] [listener]: I heard: [Hello World: 20]
[INFO] [1746197286.094652458] [listener]: I heard: [Hello World: 21]
[INFO] [1746197287.094624047] [listener]: I heard: [Hello World: 22]
[INFO] [1746197288.095175053] [listener]: I heard: [Hello World: 23]
[INFO] [1746197289.094308607] [listener]: I heard: [Hello World: 24]
[INFO] [1746197290.094362325] [listener]: I heard: [Hello World: 25]
[INFO] [1746197291.094523821] [listener]: I heard: [Hello World: 26]
[INFO] [1746197292.094526366] [listener]: I heard: [Hello World: 27]
[INFO] [1746197293.094381711] [listener]: I heard: [Hello World: 28]
[INFO] [1746197294.094353169] [listener]: I heard: [Hello World: 29]
[INFO] [1746197295.094704232] [listener]: I heard: [Hello World: 30]
```

closed 2025-05-05 by fujitatomoya