ros2/kilted_tutorial_party #303: Installation instructions

tfoote, 2025-05-01:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Setup
- Rmw: Zenoh
- BuildType: Source
- Chip: Amd64
- Os: Windows

## Links
- [Installation instructions](https://docs.ros.org/en/kilted/Installation.html)

## Checks
- [ ] **Install works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Testing this

Amronos, 2025-05-02:
Hello! I would like to work on this ticket. Can it please be assigned to me?

Innomer, 2025-05-07:
## Checks
- [x] Install works

[image]

## Result
- SUCCESS

ahcorde, 2025-05-13:
Thank you for testing

closed 2025-05-13 by ahcorde