ros2/kilted_tutorial_party #306: Intra-process executables

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Intra-process demos](https://github.com/ros2/demos/blob/kilted/intra_process_demo/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-04:
It works for me.

The files can be found here: https://github.com/suchetanrs/kilted_tutorial_party/tree/main/scripts

The full log output is below:

```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# uname -m
x86_64
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@cluttered:~# ros2 doctor --report | grep middleware
middleware name    : rmw_cyclonedds_cpp
root@cluttered:~# ./scripts/intra_process_executables.sh 

==================================== Group: two_node_pipeline =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: two_node_pipeline

Press…Published message with value: 0, and address: 0x568C9EFD0780
 Received message with value: 0, and address: 0x568C9EFD0780
Published message with value: 1, and address: 0x568C9EFD0780
 Received message with value: 1, and address: 0x568C9EFD0780
Published message with value: 2, and address: 0x568C9EFD0780
 Received message with value: 2, and address: 0x568C9EFD0780
Published message with value: 3, and address: 0x568C9EFD0780
 Received message with value: 3, and address: 0x568C9EFD0780

[ros2run]: Received signal:  Terminated
[INFO] [1746337007.757955873] [rclcpp]: signal_handler(signum=15)

==================================== Group: cyclic_pipeline =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: cyclic_pipeline

Press…Published first message with value:  42, and address: 0x5FEE5755FA00
Received message with value:         42, and address: 0x5FEE5755FA00
  sleeping for 1 second...
  done.
Incrementing and sending with value: 43, and address: 0x5FEE5755FA00
Received message with value:         43, and address: 0x5FEE5755FA00
  sleeping for 1 second...
  done.
Incrementing and sending with value: 44, and address: 0x5FEE5755FA00
Received message with value:         44, and address: 0x5FEE5755FA00
  sleeping for 1 second...
  done.
Incrementing and sending with value: 45, and address: 0x5FEE5755FA00
Received message with value:         45, and address: 0x5FEE5755FA00
  sleeping for 1 second...
  done.
Incrementing and sending with value: 46, and address: 0x5FEE5755FA00
Received message with value:         46, and address: 0x5FEE5755FA00
  sleeping for 1 second...

[ros2run]: Received signal:  Terminated
[INFO] [1746337013.442947062] [rclcpp]: signal_handler(signum=15)

==================================== Group: image_pipeline_all_in_one =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: image_pipeline_all_in_one

Press…glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau

[ros2run]: Received signal:  Terminated
[INFO] [1746337032.739224931] [rclcpp]: signal_handler(signum=15)

==================================== Group: all separately =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: camera_node
launching: watermark_node
launching: image_view_node

Press…glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau

[ros2run]: Received signal:  Terminated
[ros2run]: Received signal:  Terminated
[ros2run]: Received signal:  Terminated
[INFO] [1746337055.853390044] [rclcpp]: signal_handler(signum=15)
[INFO] [1746337055.853441817] [rclcpp]: signal_handler(signum=15)
[INFO] [1746337055.853442749] [rclcpp]: signal_handler(signum=15)

==================================== Group: two image views =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: image_pipeline_with_two_image_view

Press…glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau

[ros2run]: Received signal:  Terminated
[INFO] [1746337071.574956060] [rclcpp]: signal_handler(signum=15)

root@cluttered:~# 

```

Screenshots below:

[image]
[image]
[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde