ros2/kilted_tutorial_party #307: Intra-process executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Intra-process demos](https://github.com/ros2/demos/blob/kilted/intra_process_demo/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-04:
It works for me

[image]

[image]

[image]

Full log output below:

```

(pixi_ros2_rolling) C:\pixi_ws>cd C:\pixi_ws

(pixi_ros2_rolling) C:\pixi_ws>pixi shell

(pixi_ros2_rolling) C:\pixi_ws>call C:\pixi_ws\ros2-windows\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."

(pixi_ros2_rolling) C:\pixi_ws>ros2 doctor --report | grep middleware
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 162: UserWarning: Fail to call NetworkReport class functions.
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PackageReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point PlatformReport fails to load: No module named 'rosdistro'
C:\pixi_ws\ros2-windows\Lib\site-packages\ros2doctor\api\__init__.py: 146: UserWarning: Report entry point RosdistroReport fails to load: No module named 'rosdistro'
middleware name    : rmw_fastrtps_cpp

(pixi_ros2_rolling) C:\pixi_ws>

(pixi_ros2_rolling) C:\pixi_ws>ros2 run intra_process_demo two_node_pipeline
Published message with value: 0, and address: 0x1D96EDE6120
 Received message with value: 0, and address: 0x1D96EDE6120
Published message with value: 1, and address: 0x1D96EDE60F0
 Received message with value: 1, and address: 0x1D96EDE60F0
Published message with value: 2, and address: 0x1D96EDE60F0
 Received message with value: 2, and address: 0x1D96EDE60F0
Published message with value: 3, and address: 0x1D96EDE6160
 Received message with value: 3, and address: 0x1D96EDE6160
Published message with value: 4, and address: 0x1D96EDE6140
 Received message with value: 4, and address: 0x1D96EDE6140
Published message with value: 5, and address: 0x1D96EDE6150
 Received message with value: 5, and address: 0x1D96EDE6150
[INFO] [1746377916.140800800] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\pixi_ws>ros2 run intra_process_demo cyclic_pipeline
Published first message with value:  42, and address: 0x2241E16DED0
Received message with value:         42, and address: 0x2241E16DED0
  sleeping for 1 second...
  done.
Incrementing and sending with value: 43, and address: 0x2241E16DED0
Received message with value:         43, and address: 0x2241E16DED0
  sleeping for 1 second...
  done.
Incrementing and sending with value: 44, and address: 0x2241E16DED0
Received message with value:         44, and address: 0x2241E16DED0
  sleeping for 1 second...
  done.
Incrementing and sending with value: 45, and address: 0x2241E16DED0
Received message with value:         45, and address: 0x2241E16DED0
  sleeping for 1 second...
[INFO] [1746377926.844223200] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\pixi_ws>ros2 run intra_process_demo image_pipeline_all_in_one
[INFO] [1746377984.961961000] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\pixi_ws># Open new terminal
'#' is not recognized as an internal or external command,
operable program or batch file.

(pixi_ros2_rolling) C:\pixi_ws>ros2 run intra_process_demo camera_node
[INFO] [1746378064.427678000] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\pixi_ws># Open new terminal
'#' is not recognized as an internal or external command,
operable program or batch file.

(pixi_ros2_rolling) C:\pixi_ws>ros2 run intra_process_demo image_pipeline_with_two_image_view
[INFO] [1746378113.210376100] [rclcpp]: signal_handler(signum=2)
[ros2run]: Received signal:  Interrupt

(pixi_ros2_rolling) C:\pixi_ws>
(pixi_ros2_rolling) C:\pixi_ws>
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde