ros2/kilted_tutorial_party #308: Intra-process executables

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Intra-process demos](https://github.com/ros2/demos/blob/kilted/intra_process_demo/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Interested

suchetanrs, 2025-05-04:
It works for me. 

❗ ❗ Eventhough running the nodes separately works, it is quite glitchy with `fastdds` in comparision to `cyclone` and `zenoh`. The image frame rate is approximately 0.1-0.2 hz. It is quite fast when they share the same memory location.

The files can be found here:

The full log output is below:
```
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# ros2 doctor --report | grep middleware
middleware name    : rmw_fastrtps_cpp
root@cluttered:~# uname -m
x86_64
root@cluttered:~# ./scripts/intra_process_executables.sh 

==================================== Group: two_node_pipeline =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: two_node_pipeline

Press…Published message with value: 0, and address: 0x600FB0A5D760
 Received message with value: 0, and address: 0x600FB0A5D760
Published message with value: 1, and address: 0x600FB0A5D780
 Received message with value: 1, and address: 0x600FB0A5D780
Published message with value: 2, and address: 0x600FB0A5D760
 Received message with value: 2, and address: 0x600FB0A5D760
Published message with value: 3, and address: 0x600FB0A5D780
 Received message with value: 3, and address: 0x600FB0A5D780
Published message with value: 4, and address: 0x600FB0A5D760
 Received message with value: 4, and address: 0x600FB0A5D760

[ros2run]: Received signal:  Terminated
[INFO] [1746335883.725202435] [rclcpp]: signal_handler(signum=15)

==================================== Group: cyclic_pipeline =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: cyclic_pipeline

Press…Published first message with value:  42, and address: 0x62FEAB7F1D50
Received message with value:         42, and address: 0x62FEAB7F1D50
  sleeping for 1 second...
  done.
Incrementing and sending with value: 43, and address: 0x62FEAB7F1D50
Received message with value:         43, and address: 0x62FEAB7F1D50
  sleeping for 1 second...
  done.
Incrementing and sending with value: 44, and address: 0x62FEAB7F1D50
Received message with value:         44, and address: 0x62FEAB7F1D50
  sleeping for 1 second...
  done.
Incrementing and sending with value: 45, and address: 0x62FEAB7F1D50
Received message with value:         45, and address: 0x62FEAB7F1D50
  sleeping for 1 second...
  done.
Incrementing and sending with value: 46, and address: 0x62FEAB7F1D50
Received message with value:         46, and address: 0x62FEAB7F1D50
  sleeping for 1 second...
  done.
Incrementing and sending with value: 47, and address: 0x62FEAB7F1D50
Received message with value:         47, and address: 0x62FEAB7F1D50
  sleeping for 1 second...

[ros2run]: Received signal:  Terminated
[INFO] [1746335890.779209180] [rclcpp]: signal_handler(signum=15)

==================================== Group: image_pipeline_all_in_one =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: image_pipeline_all_in_one

Press…glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau

[ros2run]: Received signal:  Terminated
[INFO] [1746335893.817337115] [rclcpp]: signal_handler(signum=15)

==================================== Group: all separately =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: camera_node
launching: watermark_node
launching: image_view_node

Press…
[ros2run]: Received signal:  Terminated
[ros2run]: Received signal:  Terminated
[ros2run]: Received signal:  Terminated
[INFO] [1746335898.104693192] [rclcpp]: signal_handler(signum=15)
[INFO] [1746335898.104695150] [rclcpp]: signal_handler(signum=15)
[INFO] [1746335898.104691795] [rclcpp]: signal_handler(signum=15)

==================================== Group: two image views =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: image_pipeline_with_two_image_view

Press…glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau

[ros2run]: Received signal:  Terminated
[INFO] [1746335901.856127213] [rclcpp]: signal_handler(signum=15)

root@cluttered:~# 
```

Screenshots:

[image]
[image]
[image]

fujitatomoya, 2025-05-13:
@suchetanrs thanks for testing, i will go ahead to close this.

closed 2025-05-13 by fujitatomoya