ros2/kilted_tutorial_party #309: Intra-process executables

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Intra-process demos](https://github.com/ros2/demos/blob/kilted/intra_process_demo/README.md)

## Checks
- [ ] **The executables in the README are present and work as expected.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2025-05-02:
Interested in !

suchetanrs, 2025-05-04:
It works for me. 

❗ ❗ Eventhough running the nodes separately works, it is quite glitchy with `fastdds` in comparision to `cyclone` and `zenoh`. The image frame rate is approximately 0.1-0.2 hz. It is quite fast when they share the same memory location.

This is similar to the RQT issue with `fastdds`: 

The files can be found here:

The full log output is below:

```
root@cluttered:~# uname -m
x86_64
root@cluttered:~# source /opt/ros/kilted/setup.bash
root@cluttered:~# export RMW_IMPLEMENTATION=rmw_zenoh_cpp
root@cluttered:~# ros2 run rmw_zenoh_cpp rmw_zenohd &
[1] 922293
root@cluttered:~# 2025-05-04T05:46:56.019639Z  INFO ThreadId(02) zenoh::net::runtime: Using ZID: 83630e7fcb4b7ce27abc6589914d16ff
2025-05-04T05:46:56.022379Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2401:4900:3314:125b:d81e:ef76:80d9:7764]:7447
2025-05-04T05:46:56.022386Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[2401:4900:3314:125b:42cb:8d4d:511c:1374]:7447
2025-05-04T05:46:56.022387Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/[fe80::74f7:e220:82fc:df7e]:7447
2025-05-04T05:46:56.022388Z  INFO ThreadId(02) zenoh::net::runtime::orchestrator: Zenoh can be reached at: tcp/192.168.43.199:7447
Started Zenoh router with id 83630e7fcb4b7ce27abc6589914d16ff
^C
root@cluttered:~# ./scripts/intra_process_executables.sh 

==================================== Group: two_node_pipeline =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: two_node_pipeline

Press…Published message with value: 0, and address: 0x56C0751A21F0
 Received message with value: 0, and address: 0x56C0751A21F0
Published message with value: 1, and address: 0x56C0751A21F0
 Received message with value: 1, and address: 0x56C0751A21F0
Published message with value: 2, and address: 0x56C0751A21F0
 Received message with value: 2, and address: 0x56C0751A21F0
Published message with value: 3, and address: 0x56C0751A21F0
 Received message with value: 3, and address: 0x56C0751A21F0
Published message with value: 4, and address: 0x56C0751A21F0
 Received message with value: 4, and address: 0x56C0751A21F0

[ros2run]: Received signal:  Terminated
[INFO] [1746337650.285684580] [rclcpp]: signal_handler(signum=15)

==================================== Group: cyclic_pipeline =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: cyclic_pipeline

Press…Published first message with value:  42, and address: 0x60248A12A450
Received message with value:         42, and address: 0x60248A12A450
  sleeping for 1 second...
  done.
Incrementing and sending with value: 43, and address: 0x60248A12A450
Received message with value:         43, and address: 0x60248A12A450
  sleeping for 1 second...
  done.
Incrementing and sending with value: 44, and address: 0x60248A12A450
Received message with value:         44, and address: 0x60248A12A450
  sleeping for 1 second...
  done.
Incrementing and sending with value: 45, and address: 0x60248A12A450
Received message with value:         45, and address: 0x60248A12A450
  sleeping for 1 second...

[... 38 lines cut ...]

==================================== Group: two image views =======================================
===========================================================================
===========================================================================
===========================================================================
===========================================================================
launching: image_pipeline_with_two_image_view

Press…glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau

[ros2run]: Received signal:  Terminated
[INFO] [1746337723.622354392] [rclcpp]: signal_handler(signum=15)

root@cluttered:~# 

```

Screenshots below:

[image]
[image]
[image]

avanmalleghem, 2025-05-04:
- [x] The executables in the README are present and work as expected.

Can be closed !

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde