ros2/kilted_tutorial_party #310: Introducing ros2 security

tfoote, 2025-05-01:
Check the documentation for the 'Introducing ros2 security' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Introducing ros2 security page](https://docs.ros.org/en/kilted/Tutorials/Advanced/Security/Introducing-ros2-security.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-03:
I am working on this

suchetanrs, 2025-05-03:
Followed https://docs.ros.org/en/kilted/Tutorials/Advanced/Security/Introducing-ros2-security.html

The entire tutorial is OK except for a small issue.
When I built `fastdds` from source, it complained about a package not being found. It is related to https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2025-03-06/42416

I changed the command 
`colcon build --symlink-install --cmake-args -DSECURITY=ON --packages-select fastrtps rmw_fastrtps_cpp rmw_fastrtps_dynamic_cpp rmw_fastrtps_shared_cpp`

to

`colcon build --symlink-install --cmake-args -DSECURITY=ON --packages-select fastdds rmw_fastrtps_cpp rmw_fastrtps_dynamic_cpp rmw_fastrtps_shared_cpp`

Everything worked fine after this. Here is the output.

[image]

Also, I think there should be a few more commands before `colcon build --symlink-install --cmake-args -DSECURITY=ON --packages-select fastdds rmw_fastrtps_cpp rmw_fastrtps_dynamic_cpp rmw_fastrtps_shared_cpp` to clone what is necessary.

Specifically, 
```
cd ~/ws/src
git clone https://github.com/ros2/rmw_fastrtps ros2/rmw_fastrtps -b kilted
git clone https://github.com/eProsima/Fast-DDS eProsima/fastrtps
cd ..
rosdep install --from-paths src --ignore-src -r -y
colcon build --symlink-install --cmake-args -DSECURITY=ON --packages-select fastdds rmw_fastrtps_cpp rmw_fastrtps_dynamic_cpp rmw_fastrtps_shared_cpp
```

fujitatomoya, 2025-05-12:
@suchetanrs thanks for testing.

this has been fixed by https://github.com/ros2/ros2_documentation/pull/5482.

closed 2025-05-12 by fujitatomoya