ros2/kilted_tutorial_party #312: Introduction To Tf2

tfoote, 2025-05-01:
Check the documentation for the 'Introduction To Tf2' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Introduction To Tf2 page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Introduction-To-Tf2.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-03:
I am working on this

suchetanrs, 2025-05-03:
There is an issue with `turtlesim` package. I think it's because the message defs have been moved to `turtlesim_msgs`

When I run `ros2 launch turtle_tf2_py turtle_tf2_demo.launch.py`
I got the following error

```root@cluttered:~# source /opt/ros/kilted/setup.bash                                                                                                     10:56:15 [4/4]
root@cluttered:~# ros2 launch turtle_tf2_py turtle_tf2_demo.launch.py                                                                                                 
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-05-03-10-55-26-404544-cluttered-1720117                                                         
[INFO] [launch]: Default logging verbosity is set to INFO                                                                                                             
[INFO] [turtlesim_node-1]: process started with pid [1720120]                                                                                                         
[INFO] [turtle_tf2_broadcaster-2]: process started with pid [1720121]
[INFO] [turtle_tf2_broadcaster-3]: process started with pid [1720122]
[INFO] [turtle_tf2_listener-4]: process started with pid [1720123]
[turtlesim_node-1] [INFO] [1746249926.508188288] [sim]: Starting turtlesim with node name /sim
[turtlesim_node-1] [INFO] [1746249926.511491880] [sim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
[turtlesim_node-1] glx: failed to create dri3 screen
[turtlesim_node-1] failed to load driver: nouveau
[turtlesim_node-1] glx: failed to create dri3 screen
[turtlesim_node-1] failed to load driver: nouveau
[turtle_tf2_listener-4] Traceback (most recent call last):
[turtle_tf2_listener-4]   File "/opt/ros/kilted/lib/turtle_tf2_py/turtle_tf2_listener", line 33, in <module>
[turtle_tf2_listener-4]     sys.exit(load_entry_point('turtle-tf2-py==0.6.3', 'console_scripts', 'turtle_tf2_listener')())
[turtle_tf2_listener-4]              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[turtle_tf2_listener-4]   File "/opt/ros/kilted/lib/turtle_tf2_py/turtle_tf2_listener", line 25, in importlib_load_entry_point
[turtle_tf2_listener-4]     return next(matches).load()
[turtle_tf2_listener-4]            ^^^^^^^^^^^^^^^^^^^^
[turtle_tf2_listener-4]   File "/usr/lib/python3.12/importlib/metadata/__init__.py", line 205, in load
[turtle_tf2_listener-4]     module = import_module(match.group('module'))
[turtle_tf2_listener-4]              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[turtle_tf2_listener-4]   File "/usr/lib/python3.12/importlib/__init__.py", line 90, in import_module
[turtle_tf2_listener-4]     return _bootstrap._gcd_import(name[level:], package, level)
[turtle_tf2_listener-4]            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[turtle_tf2_listener-4]   File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
[turtle_tf2_listener-4]   File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
[turtle_tf2_listener-4]   File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
[turtle_tf2_listener-4]   File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
[turtle_tf2_listener-4]   File "<frozen importlib._bootstrap_external>", line 995, in exec_module
[turtle_tf2_listener-4]   File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
[turtle_tf2_listener-4]   File "/opt/ros/kilted/lib/python3.12/site-packages/turtle_tf2_py/turtle_tf2_listener.py", line 27, in <module>
[turtle_tf2_listener-4]     from turtlesim.srv import Spawn
[turtle_tf2_listener-4] ModuleNotFoundError: No module named 'turtlesim'
[ERROR] [turtle_tf2_listener-4]: process has died [pid 1720123, exit code 1, cmd '/opt/ros/kilted/lib/turtle_tf2_py/turtle_tf2_listener --ros-args -r __node:=listener --params-file /tmp/launch_params_4xi81uua'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```

suchetanrs, 2025-05-03:
I built the broken package after editing the line from source and it worked fine.

[image]

Frames works too

[image]

❗ ❗  **There is an issue with rviz**

```
root@cluttered:~# ros2 run rviz2 rviz2 -d $(ros2 pkg prefix --share turtle_tf2_py)/rviz/turtle_rviz.rviz
glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau
glx: failed to create dri3 screen
failed to load driver: nouveau
[INFO] [1746250584.794075345] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1746250584.794124224] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[INFO] [1746250584.808657332] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'resource_retriever::Exception'
  what():  Error retrieving file [/opt/ros/kilted/share/rviz_default_plugins/icons/classes/TF.svg]: Failed to open file
[ros2run]: Aborted
root@cluttered:~# 
```

suchetanrs, 2025-05-03:
PR for the incorrect import is raised here https://github.com/ros/geometry_tutorials/pull/88

Aniketkumarroy, 2025-05-09:
[image]

the rviz2 visualization is also working fine after the latest fix and updating te packages(sudo apt update && sudo apt upgrade). all other parts of this documentation are tested

closed 2025-05-09 by ahcorde