ros2/kilted_tutorial_party #313: Latest Development Setup

tfoote, 2025-05-01:
Check the documentation for the 'Latest Development Setup' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Latest Development Setup page](https://docs.ros.org/en/kilted/Installation/Alternatives/Latest-Development-Setup.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Tanishq30052002, 2025-05-03:
I am testing it out

Tanishq30052002, 2025-05-03:
```
vcs import --input https://raw.githubusercontent.com/ros2/ros2/kilted/ros2.repos src
bash: vcs: command not found
```

I have to run `sudo apt install python3-vcstool` to install vcs.
I think we should add it to the documentation

Tanishq30052002, 2025-05-03:
```
root@4b54f41ec001:~/ros2_kilted# rosdep init
bash: rosdep: command not found
```
I have to run sudo apt install python3-rosdep to install rosdep.
I think we should add it to the documentation

Tanishq30052002, 2025-05-03:
```
root@4b54f41ec001:~/ros2_kilted# colcon build --symlink-install
bash: colcon: command not found
```
I have to run `sudo apt install python3-colcon-common-extensions` to install colcon.
I think we should add it to the documentation

Tanishq30052002, 2025-05-03:
After having these changes everything was working fine.

christophebedard, 2025-05-03:
As mentioned in https://github.com/ros2/ros2_documentation/pull/5494#pullrequestreview-2813305077, these should all be installed when you install `ros-dev-tools`. Can you make sure you're correctly installing `ros-dev-tools`?

Tanishq30052002, 2025-05-03:
> As mentioned in [ros2/ros2_documentation#5494 (review)](https://github.com/ros2/ros2_documentation/pull/5494#pullrequestreview-2813305077), these should all be installed when you install `ros-dev-tools`. Can you make sure you're correctly installing `ros-dev-tools`?

@christophebedard I apologize, I tried a fresh install and it works. Might be I missed something earlier.
I will close the issue on documentation.
Thanks

Here now, everything works fine.

christophebedard, 2025-05-03:
Great, thanks for checking again!

closed 2025-05-03 by christophebedard