ros2/kilted_tutorial_party #314: Launch testing

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Launch testing examples](https://github.com/ros2/launch/blob/kilted/launch_testing/README.md#examples)

## Checks
- [ ] **Clone the example repository**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       mkdir -p ~/launch_ws/src
       ```
    1.
       ```bash
       # User input in terminal 1
       git clone https://github.com/ros2/launch -b iron ~/launch_ws/src/launch
       ```
  
    
  </details>
- [ ] **Test `hello_world_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/hello_world_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `good_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/good_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `terminating_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/terminating_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `args_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Show the arguments
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py --show-args
       ```
    1.
       Run the test
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py dut_arg:=hey
       ```
  
    
  </details>
- [ ] **Test `context_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/context_launch_test.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Malay-dev, 2025-05-05:
### Environment Setup

```
Operating System: Windows
Rmw: Cyclone
BuildType: Binary
```
### ROS Doctor report
```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19045-SP0
release          : 10
processor        : Intel64 Family 6 Model 140 Stepping 1, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : kilted
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0
```

### Testing Setup
- [x]  Clone the example repo

```
cd C:\pixi_ws
git clone https://github.com/ros2/launch -b iron launch_ws\src\launch
```

Tests

- [x] Test `hello_world_launch_test.py`
```
(pixi_ros2_rolling) C:\pixi_ws>launch_test launch_ws/src/launch/launch_testing/test/launch_testing/examples/hello_world_launch_test.py
[INFO] [launch]: All log files can be found below C:\Users\Malay\.ros\log\2025-05-05-19-34-26-075662-LAPTOP-DUTVF5EE-10736
[INFO] [launch]: Default logging verbosity is set to INFO
test_read_stdout (hello_world_launch_test.TestHelloWorldProcess.test_read_stdout)
Check if 'hello_world' was found in the stdout. ... [INFO] [echo-1]: process started with pid [2000]
ok

----------------------------------------------------------------------
Ran 1 test in 0.018s

OK
[INFO] [echo-1]: process has finished cleanly [pid 2000]
test_exit_codes (hello_world_launch_test.TestHelloWorldShutdown.test_exit_codes)
Check if the processes exited normally. ... ok

----------------------------------------------------------------------
Ran 1 test in 0.000s

OK   
``` 

- [x] Test good_proc_launch_test.py
```
(pixi_ros2_rolling) C:\pixi_ws>launch_test launch_ws/src/launch/launch_testing/test/launch_testing/examples/good_proc_launch_test.py
[INFO] [launch]: All log files can be found below C:\Users\Malay\.ros\log\2025-05-05-19-38-02-604007-LAPTOP-DUTVF5EE-15648
[... 122 lines cut ...]
[INFO] [launch]: Default logging verbosity is set to INFO
test_process_output (context_launch_test.TestProcOutput.test_process_output) ... [INFO] [good_proc.py-1]: process started with pid [7604]
ok

----------------------------------------------------------------------
Ran 1 test in 1.029s

OK
[WARNING] [good_proc.py-1]: 'SIGINT' sent to process[good_proc.py-1] not supported on Windows, escalating to 'SIGTERM'
[INFO] [good_proc.py-1]: sending signal 'SIGTERM' to process[good_proc.py-1]
[ERROR] [good_proc.py-1]: process has died [pid 7604, exit code 1, cmd 'C:\pixi_ws\.pixi\envs\default\python.exe C:\pixi_ws\ros2-windows\lib/launch_testing\good_proc.py'].
test_all_context_objects (context_launch_test.TestProcessOutput.test_all_context_objects) ... ok
test_all_context_objects_different_order (context_launch_test.TestProcessOutput.test_all_context_objects_different_order) ... ok
test_full_output (context_launch_test.TestProcessOutput.test_full_output) ... ok
test_int_val (context_launch_test.TestProcessOutput.test_int_val) ... ok

----------------------------------------------------------------------
Ran 4 tests in 0.001s

OK
```

Ref: Similar results as #316 

Here is the testing report @tfoote @ahcorde

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde