ros2/kilted_tutorial_party #315: Launch testing

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Launch testing examples](https://github.com/ros2/launch/blob/kilted/launch_testing/README.md#examples)

## Checks
- [ ] **Clone the example repository**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       mkdir -p ~/launch_ws/src
       ```
    1.
       ```bash
       # User input in terminal 1
       git clone https://github.com/ros2/launch -b iron ~/launch_ws/src/launch
       ```
  
    
  </details>
- [ ] **Test `hello_world_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/hello_world_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `good_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/good_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `terminating_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/terminating_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `args_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Show the arguments
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py --show-args
       ```
    1.
       Run the test
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py dut_arg:=hey
       ```
  
    
  </details>
- [ ] **Test `context_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/context_launch_test.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Hi, I am working on this

4-legends, 2025-05-04:
- Setup
```
legends@4legends-ASUS:~$ printenv | grep RMW
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

legends@4legends-ASUS:~$ printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=kilted

``` 

- [x] Clone the example repository
- [x] Test hello_world_launch_test.py

```
legends@4legends-ASUS:~/ros2_ws$ launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/hello_world_launch_test.py

1746326470.3895063 [INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-03-19-41-10-387483-4legends-ASUS-28010
1746326470.3896863 [INFO] [launch]: Default logging verbosity is set to INFO
1746326470.3918648 [INFO] [echo-1]: process started with pid [28012]
1746326470.3924179 [echo-1] hello_world
1746326470.3931000 [INFO] [echo-1]: process has finished cleanly [pid 28012]
``` 

- [x] Test good_proc_launch_test.py

```
legends@4legends-ASUS:~/ros2_ws$ launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/good_proc_launch_test.py
[INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-03-19-43-41-997234-4legends-ASUS-28137
[INFO] [launch]: Default logging verbosity is set to INFO
test_count_to_four (good_proc_launch_test.TestGoodProcess.test_count_to_four) ... [INFO] [python3-1]: process started with pid [28142]
[python3-1] Starting Up
[python3-1] Loop 0
[python3-1] Loop 1
[python3-1] Loop 2
[python3-1] Loop 3
[python3-1] Loop 4
ok

----------------------------------------------------------------------
Ran 1 test in 4.060s

OK
[INFO] [python3-1]: sending signal 'SIGINT' to process[python3-1]
[python3-1] Shutting Down
[INFO] [python3-1]: process has finished cleanly [pid 28142]
test_exit_code (good_proc_launch_test.TestProcessOutput.test_exit_code) ... ok
test_full_output (good_proc_launch_test.TestProcessOutput.test_full_output) ... ok
test_out_of_order (good_proc_launch_test.TestProcessOutput.test_out_of_order) ... ok

----------------------------------------------------------------------
Ran 3 tests in 0.001s

OK
``` 

- [x] Test terminating_proc_launch_test.py
```
legends@4legends-ASUS:~/ros2_ws$ launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/terminating_proc_launch_test.py
[INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-03-19-45-06-939338-4legends-ASUS-28177
[INFO] [launch]: Default logging verbosity is set to INFO
test_no_args (terminating_proc_launch_test.TestTerminatingProc.test_no_args)
Test terminating_proc without command line arguments. ... [INFO] [terminating_proc.py-1]: process started with pid [28182]
[terminating_proc.py-1] Starting Up
[terminating_proc.py-1] Ready
[terminating_proc.py-1] Emulating Work
[terminating_proc.py-1] Done
[... 70 lines cut ...]
Ran 1 test in 1.022s

OK
[INFO] [good_proc.py-1]: sending signal 'SIGINT' to process[good_proc.py-1]
[INFO] [good_proc.py-1]: process has finished cleanly [pid 28705]
test_all_context_objects (context_launch_test.TestProcessOutput.test_all_context_objects) ... ok
test_all_context_objects_different_order (context_launch_test.TestProcessOutput.test_all_context_objects_different_order) ... ok
test_full_output (context_launch_test.TestProcessOutput.test_full_output) ... ok
test_int_val (context_launch_test.TestProcessOutput.test_int_val) ... ok

----------------------------------------------------------------------
Ran 4 tests in 0.000s

OK

1746326998.8190186 [INFO] [launch]: All log files can be found below /home/legends/.ros/log/2025-05-03-19-49-58-816378-4legends-ASUS-28699
1746326998.8191788 [INFO] [launch]: Default logging verbosity is set to INFO
1746326998.8212266 [INFO] [good_proc.py-1]: process started with pid [28705]
1746326998.8417809 [good_proc.py-1] Starting Up
1746326998.8418705 [good_proc.py-1] Loop 0
1746326999.8420913 [good_proc.py-1] Loop 1
1746326999.8448656 [INFO] [good_proc.py-1]: sending signal 'SIGINT' to process[good_proc.py-1]
1746326999.8454921 [good_proc.py-1] Shutting Down
1746326999.8531642 [INFO] [good_proc.py-1]: process has finished cleanly [pid 28705]
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde