ros2/kilted_tutorial_party #317: Launch testing

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Launch testing examples](https://github.com/ros2/launch/blob/kilted/launch_testing/README.md#examples)

## Checks
- [ ] **Clone the example repository**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       mkdir -p ~/launch_ws/src
       ```
    1.
       ```bash
       # User input in terminal 1
       git clone https://github.com/ros2/launch -b iron ~/launch_ws/src/launch
       ```
  
    
  </details>
- [ ] **Test `hello_world_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/hello_world_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `good_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/good_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `terminating_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/terminating_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `args_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Show the arguments
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py --show-args
       ```
    1.
       Run the test
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py dut_arg:=hey
       ```
  
    
  </details>
- [ ] **Test `context_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/context_launch_test.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

NickTziaros, 2025-05-03:
## Setup
Used the ros-testing:kilted-desktop image.
[image]

## Summary

✅ As expected

## Clone the example repository:

`mkdir -p ~/launch_ws/src`

I assume that I am supposed to clone from the kilted branch and not Iron. 

Instead of 

`git clone https://github.com/ros2/launch -b iron ~/launch_ws/src/launch`

I run 

`git clone https://github.com/ros2/launch -b kilted ~/launch_ws/src/launch`

## Test `hello_world_launch_test.py` : 

- `launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/hello_world_launch_test.py`

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-05-03-10-38-01-007405-99b0e257db86-865
[INFO] [launch]: Default logging verbosity is set to INFO
test_read_stdout (hello_world_launch_test.TestHelloWorldProcess.test_read_stdout)
Check if 'hello_world' was found in the stdout. ... [INFO] [echo-1]: process started with pid [867]
[INFO] [echo-1]: process has finished cleanly [pid 867]
ok

----------------------------------------------------------------------
Ran 1 test in 0.002s

OK
test_exit_codes (hello_world_launch_test.TestHelloWorldShutdown.test_exit_codes)
Check if the processes exited normally. ... ok

----------------------------------------------------------------------
Ran 1 test in 0.000s

OK
```
## Test `good_proc_launch_test.py` :

- `launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/good_proc_launch_test.py`

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-05-03-10-43-53-352262-99b0e257db86-3981
[INFO] [launch]: Default logging verbosity is set to INFO
test_count_to_four (good_proc_launch_test.TestGoodProcess.test_count_to_four) ... [INFO] [python3-1]: process started with pid [3990]
[python3-1] Starting Up
[python3-1] Loop 0
[python3-1] Loop 1
[python3-1] Loop 2
[python3-1] Loop 3
[python3-1] Loop 4
ok

----------------------------------------------------------------------
Ran 1 test in 4.022s

OK
[INFO] [python3-1]: sending signal 'SIGINT' to process[python3-1]
[python3-1] Shutting Down
[INFO] [python3-1]: process has finished cleanly [pid 3990]
test_exit_code (good_proc_launch_test.TestProcessOutput.test_exit_code) ... ok
[... 98 lines cut ...]

- `launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/context_launch_test.py`

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-05-03-10-49-59-841364-99b0e257db86-7359
[INFO] [launch]: Default logging verbosity is set to INFO
test_process_output (context_launch_test.TestProcOutput.test_process_output) ... [INFO] [good_proc.py-1]: process started with pid [7361]
ok

----------------------------------------------------------------------
Ran 1 test in 1.018s

OK
[INFO] [good_proc.py-1]: sending signal 'SIGINT' to process[good_proc.py-1]
[INFO] [good_proc.py-1]: process has finished cleanly [pid 7361]
test_all_context_objects (context_launch_test.TestProcessOutput.test_all_context_objects) ... ok
test_all_context_objects_different_order (context_launch_test.TestProcessOutput.test_all_context_objects_different_order) ... ok
test_full_output (context_launch_test.TestProcessOutput.test_full_output) ... ok
test_int_val (context_launch_test.TestProcessOutput.test_int_val) ... ok

----------------------------------------------------------------------
Ran 4 tests in 0.001s

OK
```

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde