ros2/kilted_tutorial_party #318: Launch testing

tfoote, 2025-05-01:
## Setup
- Rmw: Zenoh
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Launch testing examples](https://github.com/ros2/launch/blob/kilted/launch_testing/README.md#examples)

## Checks
- [ ] **Clone the example repository**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       mkdir -p ~/launch_ws/src
       ```
    1.
       ```bash
       # User input in terminal 1
       git clone https://github.com/ros2/launch -b iron ~/launch_ws/src/launch
       ```
  
    
  </details>
- [ ] **Test `hello_world_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/hello_world_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `good_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/good_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `terminating_proc_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/terminating_proc_launch_test.py
       ```
  
    
  </details>
- [ ] **Test `args_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       Show the arguments
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py --show-args
       ```
    1.
       Run the test
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/args_launch_test.py dut_arg:=hey
       ```
  
    
  </details>
- [ ] **Test `context_launch_test.py`**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       ```bash
       # User input in terminal 1
       launch_test ~/launch_ws/src/launch/launch_testing/test/launch_testing/examples/context_launch_test.py
       ```
  
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-02:
I have tested all the following, and everything is working as expected.

## Checks

* [x]  **Clone the example repository**   
* [x]  **Test `hello_world_launch_test.py`**   

[image]

* [x]  **Test `good_proc_launch_test.py`**   

[image]

* [x]  **Test `terminating_proc_launch_test.py`**   

[image]

* [x]  **Test `args_launch_test.py`**   

[image]

* [x]  **Test `context_launch_test.py`**   

[image]

ahcorde, 2025-05-02:
Thank you for testing

closed 2025-05-02 by ahcorde