ros2/kilted_tutorial_party #320: Learning About Tf2 And Time Py

tfoote, 2025-05-01:
Check the documentation for the 'Learning About Tf2 And Time Py' page

## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Learning About Tf2 And Time Py page](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Learning-About-Tf2-And-Time-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

4-legends, 2025-05-04:
The Link is not valid

https://docs.ros.org/en/kilted/Tutorials/Intermediate/Tf2/Learning-About-Tf2-And-Time-Py.html

Not Found

The requested URL was not found on this server.
Apache Server at docs.ros.org Port 443

JatinVira, 2025-05-05:
@ahcorde 

There is no page for both `Time and tf2` and `travelling in time` for Python counterparts on the `Kilted` tutorial page under `Intermediate` tab

[image]

Yadunund, 2025-05-05:
Thanks for flagging this issue. The test case was incorrectly generated for a tutorial that does not exist. I've pushed https://github.com/osrf/ros2_test_cases/pull/1560/commits/4e797d23bedc5cd8d4c4177a7bd16b621e09bf63 to fix the test case generation for the next iteration.

closed 2025-05-05 by Yadunund

JatinVira, 2025-05-06:
Can this issue have an assigne? @Yadunund