ros2/kilted_tutorial_party #321: Lifecycle demo

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/kilted/lifecycle/README.rst)

## Checks
- [ ] **Run files individually**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       [image]
       ```bash
       # User input in terminal 1
       ros2 run lifecycle lifecycle_talker
       ```
    1.
       [image]
       ```bash
       # User input in terminal 2
       ros2 run lifecycle lifecycle_listener
       ```
    1.
       [image]
       ```bash
       # User input in terminal 3
       ros2 run lifecycle lifecycle_service_client
       ```
  
    **Expect**
    
    1.
       The images match.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

ShahazadAbdulla, 2025-05-05:
I'll test this on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS

ShahazadAbdulla, 2025-05-05:
Successfully completed the check for Issue #321 on Windows 11 / amd64 / binary (via Pixi) / CycloneDDS.

- Ran `ros2 run lifecycle_py lifecycle_talker` in T1, `ros2 run lifecycle_py lifecycle_listener` in T2, and `ros2 run lifecycle_py lifecycle_service_client` in T3 simultaneously.
- Observed the output across all three terminals.
- The `lifecycle_service_client` successfully triggered all expected state transitions (configure, activate, deactivate, cleanup, shutdown) on the `lifecycle_talker`.
- The `lifecycle_talker` executed the corresponding `on_configure`, `on_activate`, etc., callbacks and only published data when in the 'active' state.
- The `lifecycle_listener` received the correct state transition notifications and only received data messages when the talker was active.

The observed behavior across all three nodes perfectly matched the sequence described in the linked lifecycle tutorial.

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde