ros2/kilted_tutorial_party #322: Lifecycle demo

tfoote, 2025-05-01:
## Setup
- Rmw: Cyclone
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/kilted/lifecycle/README.rst)

## Checks
- [ ] **Run files individually**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       [image]
       ```bash
       # User input in terminal 1
       ros2 run lifecycle lifecycle_talker
       ```
    1.
       [image]
       ```bash
       # User input in terminal 2
       ros2 run lifecycle lifecycle_listener
       ```
    1.
       [image]
       ```bash
       # User input in terminal 3
       ros2 run lifecycle lifecycle_service_client
       ```
  
    **Expect**
    
    1.
       The images match.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

khaledgabr77, 2025-05-03:
I have completed the following tests and confirmed everything works as expected.

## Checks

* [x]  **Run files individually**   

[image]

ahcorde, 2025-05-05:
Thank you for testing

closed 2025-05-05 by ahcorde