ros2/kilted_tutorial_party #323: Lifecycle demo

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Binary
- Os: Windows
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/kilted/lifecycle/README.rst)

## Checks
- [ ] **Run files individually**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       [image]
       ```bash
       # User input in terminal 1
       ros2 run lifecycle lifecycle_talker
       ```
    1.
       [image]
       ```bash
       # User input in terminal 2
       ros2 run lifecycle lifecycle_listener
       ```
    1.
       [image]
       ```bash
       # User input in terminal 3
       ros2 run lifecycle lifecycle_service_client
       ```
  
    **Expect**
    
    1.
       The images match.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Poo-09, 2025-05-05:
Interested, Please assign this to me.

Poo-09, 2025-05-06:
[image]

[image]

[image]

the lifecycle_talker and liecycle_publisher does not get active until this command is not run "# User input in terminal 3
ros2 run lifecycle lifecycle_service_client"

when we run lifecycle_sevice_client then the lifecycle_talker gets activated and start sending messages

Tanishq30052002, 2025-05-13:
https://github.com/user-attachments/assets/46489067-a883-449c-a8d4-963c0cfb0e2a
@ahcorde Tested the working.
Working as expected

ahcorde, 2025-05-14:
Thank you for testing

closed 2025-05-14 by ahcorde