ros2/kilted_tutorial_party #324: Lifecycle demo

tfoote, 2025-05-01:
## Setup
- Rmw: Fastdds
- BuildType: Debian
- Os: Noble
- Chip: Amd64

## Links
- [Reference URL](https://github.com/ros2/demos/blob/kilted/lifecycle/README.rst)

## Checks
- [ ] **Run files individually**

     

  <details><summary>details</summary>
  
    **Try**
  
    1.
       [image]
       ```bash
       # User input in terminal 1
       ros2 run lifecycle lifecycle_talker
       ```
    1.
       [image]
       ```bash
       # User input in terminal 2
       ros2 run lifecycle lifecycle_listener
       ```
    1.
       [image]
       ```bash
       # User input in terminal 3
       ros2 run lifecycle lifecycle_service_client
       ```
  
    **Expect**
    
    1.
       The images match.
    
  </details>

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

NickTziaros, 2025-05-02:
## Setup
Used the ros-testing:kilted-desktop image.
[image]

## Comments:
 The demo works as expected with a minor difference in the ``ros2 run lifecycle lifecycle_talker`` output where the warning appears only once when the publisher is not active.

**My output:** 
[image]
**Demo output:**
[image]

### ❓   **Is this statement true?** 

> By default, the on_error returns CallbackReturn::FAILURE and the state machine transitions into finalized.

 In [rclcpp](https://github.com/ros2/rclcpp/blob/kilted/rclcpp_lifecycle/src/node_interfaces/lifecycle_node_interface.cpp) the default of ``on_error()`` seems to be returning SUCCESS.

## Output : 

### Demo

`ros2 run lifecycle lifecycle_talker`

[image]

`ros2 run lifecycle lifecycle_listener`

[image]

`ros2 run lifecycle lifecycle_service_client`

[image]

### CLI :

[image]

### Services :

[image]

fujitatomoya, 2025-05-10:
> In [rclcpp](https://github.com/ros2/rclcpp/blob/kilted/rclcpp_lifecycle/src/node_interfaces/lifecycle_node_interface.cpp) the default of on_error() seems to be returning SUCCESS.

yes this needs to be implemented by user. it always returns SUCCESS in default.

closed 2025-05-10 by fujitatomoya