ros2/kilted_tutorial_party #326: Local workspace overlay

tfoote, 2025-05-01:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Setup
- Rmw: Connext
- BuildType: Binary
- Chip: Amd64
- Os: Noble

## Checks
- [ ] **Local workspace overlay works**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

suchetanrs, 2025-05-02:
Hi, I worked on this ticket

I built bondcpp from source (https://github.com/ros/bond_core) which depends on rclcpp and rclcpp_lifecycle.

```
root@cluttered:~/custom_ws# export RMW_IMPLEMENTATION=rmw_connextdds
root@cluttered:~/custom_ws# colcon build --symlink-install
Starting >>> bond    
Starting >>> smclib
Finished <<< smclib [0.89s]                                                  
Finished <<< bond [3.90s]                    
Starting >>> bondcpp
Starting >>> bondpy
Finished <<< bondpy [0.92s]                                                 
Finished <<< bondcpp [7.82s]                      
Starting >>> bond_core
Starting >>> test_bond                   
Finished <<< bond_core [0.56s]                                                        
Finished <<< test_bond [10.3s]                      

Summary: 6 packages finished [22.1s]
root@cluttered:~/custom_ws# source install/setup.bash
root@cluttered:~/custom_ws# colcon test --packages-up-to bondcpp
Starting >>> bond    
Starting >>> smclib
Finished <<< smclib [2.95s]                                    
Finished <<< bond [5.11s]          
Starting >>> bondcpp
Finished <<< bondcpp [1.36s]          

Summary: 3 packages finished [6.58s]
root@cluttered:~/custom_ws# ros2 doctor --report | grep middleware
middleware name    : rmw_connextdds
root@cluttered:~/custom_ws# uname -m
x86_64
root@cluttered:~/custom_ws# colcon test-result
Summary: 242 tests, 0 errors, 0 failures, 59 skipped
root@cluttered:~/custom_ws# 

```

ahcorde, 2025-05-02:
Thank you for testing!